adafruit/Adafruit_CircuitPython_ESP32SPI #5: failure to fetch data results in looping failures

jerryneedell, 2019-02-18:
While running the esp32spi_cheerlights.py example I have occasionally found it stuck in a loop of failures at https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/examples/esp32spi_cheerlights.py#L49

It typically runs for several hours before this failure occurs.

A simple workaround appears to be to add an 
```
esp.reset()
```
when this error is detected - I am running a long test to see if I can catch the error and verify that it recovers.  adding the esp.reset() before every "get" as a test seemed to work OK.

If this works - I'll put in PR's for the examples

makermelissa, 2019-02-18:
What if you add the reset between the print and continue lines?

jerryneedell, 2019-02-18:
that is what I have done.
```
import time
import board
import busio
from digitalio import DigitalInOut

from adafruit_esp32spi import adafruit_esp32spi
from adafruit_esp32spi import adafruit_esp32spi_wifimanager

import neopixel
import adafruit_fancyled.adafruit_fancyled as fancy

# Get wifi details and more from a settings.py file
try:
    from esp32spi_settings import settings
except ImportError:
    print("WiFi settings are kept in esp32spi_settings.py, please add them there!")
    raise

print("ESP32 SPI webclient test")

DATA_SOURCE = "https://api.thingspeak.com/channels/1417/feeds.json?results=1"
DATA_LOCATION = ["feeds", 0, "field2"]

esp32_cs = DigitalInOut(board.D9)
esp32_ready = DigitalInOut(board.D10)
esp32_reset = DigitalInOut(board.D5)
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(esp, settings, board.NEOPIXEL)

# neopixels
pixels = neopixel.NeoPixel(board.A1, 16, brightness=0.3)
pixels.fill(0)

# we'll save the value in question
last_value = value = None

while True:
    try:
        print("Fetching json from", DATA_SOURCE)
        response = wifi.get(DATA_SOURCE)
        print(response.json())
        value=response.json()
        for key in DATA_LOCATION:
            value = value[key]
            print(value)
        response.close()
    except (ValueError, RuntimeError) as e:
        print("Failed to get data, retrying\n", e)
        print("RESET ESP")
        esp.reset()
        continue

    if not value:
        continue
    if last_value != value:
        color = int(value[1:],16)
        red = color >> 16 & 0xFF
        green = color >> 8 & 0xFF
        blue = color& 0xFF
        gamma_corrected = fancy.gamma_adjust(fancy.CRGB(red, green, blue)).pack()

        pixels.fill(gamma_corrected)
        last_value = value
    response = None
    time.sleep(60)
```

jerryneedell, 2019-02-18:
adding it on every get was just a test -- not meant for a fix

makermelissa, 2019-02-18:
Oh I get it now. Thanks. I just meant test the modification in the library itself to see if that does fix it.

jerryneedell, 2019-02-18:
I'm not sure we need to force the RESET in the library -- it may be better to let the user decide how to respond.= as in the example.

jerryneedell, 2019-02-18:
OK -- after running fo over 12 hours, it did catch an error and the RESET was successful
```
Fetching json from https://api.thingspeak.com/channels/1417/feeds.json?results=1
Failed to get data, retrying
 Expected 01 but got 00
RESET ESP
Fetching json from https://api.thingspeak.com/channels/1417/feeds.json?results=1
{'channel': {'id': 1417, 'latitude': '40.5', 'name': 'CheerLights', 'longitude': '-80.22', 'updated_at': '2018-07-20T01:05:15Z', 'description': 'CheerLights is an \u201cInternet of Things\u201d project created by Hans Scharler that allows people\u2019s lights all across the world to synchronize to one color set by Twitter. This is a way to connect physical things with social networking experiences.', 'last_entry_id': 533080, 'field2': 'CheerLights HEX Color', 'field1': 'Last CheerLights Command', 'created_at': '2011-12-01T03:30:04Z'}, 'feeds': [{'created_at': '2019-02-18T13:38:10Z', 'field2': '#0000FF', 'field1': 'blue', 'entry_id': 533080}]}
[{'created_at': '2019-02-18T13:38:10Z', 'field2': '#0000FF', 'field1': 'blue', 'entry_id': 533080}]
{'created_at': '2019-02-18T13:38:10Z', 'field2': '#0000FF', 'field1': 'blue', 'entry_id': 533080}
#0000FF
```

Should I go ahead and add this to the examples that need it or do you want to implement it differently?

makermelissa, 2019-02-18:
I want to implement it a little differently. I’ll go ahead and get that changed this morning. Thanks for finding this.

ladyada, 2019-02-18:
hiya i'd like the wifimanager to hard reset on communication failure at the top of the connection-check section so in theory the exception is caught, it goes to the next loop where it tries to fetch data, and hard-resets there

jerryneedell, 2019-02-18:
It was not clear to me if the communication failure I was seeing resulted in a loss of connection or not. It just kept retrying the get and was failing the get_host_by_name.  Issuing the reset after the failed get attempt forced it to reconnect at the next loop but it was done outside the wifi-manager.

makermelissa, 2019-02-18:
I created a PR that will hopefully address this.

makermelissa, 2019-02-19:
Closed #5 via #6.

closed 2019-02-19 by makermelissa