adafruit/Adafruit_CircuitPython_ESP32SPI #8: Linking to or including ESP32 Firmware

makermelissa, 2019-02-19:
I'm wondering if we should include the ESP32 firmware in the examples (along with instructions) or maybe even just link to it. Currently it's available at https://github.com/ladyada/CircuitPython_WiFi_Demos/blob/master/esp32program/NINA_W102.bin

and the command to flash it (on a Mac at least) is:
`/usr/local/bin/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0 NINA_W102.bin`

Of course `/dev/ttyUSB0` is bound to be different on different systems such as `COM6` on windows. I believe there's a command to display this, but I couldn't find it just now. I think updating the guide (which it mentions should be soon) would take care of the instructions, but I'm wondering if we should include the firmware in the examples.

ladyada, 2019-02-20:
i think best if we add it to the existing ESP co-processor guide and link there?

makermelissa, 2019-02-21:
Yeah, that would probably work best.

makermelissa, 2019-02-23:
Looks like the guide was updated with this. Closing.

closed 2019-02-23 by makermelissa