adafruit/Adafruit_CircuitPython_ESP32SPI #9: json error in get

jerryneedell, 2019-02-20:
I was trying to test @brentru  adafruit_io library and started getting jason errors when trying to use his "get_all_feeds" function  -- I then tried the same "get" just using esp32spi (via wifimanager) and get the same error so there seems to be an underlying issue in esp32spi

here is the test case
```
import time
import board
import busio
from digitalio import DigitalInOut

from adafruit_esp32spi import adafruit_esp32spi
from adafruit_esp32spi import adafruit_esp32spi_wifimanager

print("ESP32 SPI webclient test")

# Get wifi details and more from a settings.py file
try:
    from esp32spi_settings import settings
except ImportError:
    print("WiFi settings are kept in esp32spi_settings.py, please add them there!")
    raise

esp32_cs = DigitalInOut(board.D9)
esp32_ready = DigitalInOut(board.D10)
esp32_reset = DigitalInOut(board.D5)
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(esp, settings, board.NEOPIXEL)
esp._debug=True
while True:
    try:
        print("Get data...", end='')
        response = wifi.get(
            "https://io.adafruit.com/api/v2/"+settings['aio_username']+"/feeds",
            headers={bytes("X-AIO-KEY","utf-8"):bytes(settings['aio_key'],"utf-8")})
        print(response.json())
        response.close()
        print("OK")
    except (ValueError, RuntimeError) as e:
        print("Failed to get data, retrying\n", e)
        wifi.reset()
        continue
    response = None
    time.sleep(15)

```
here is the log -- with some repeated lines removed

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-97-g0dc260058 on 2019-02-19; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> import aio_get_esp32spi
ESP32 SPI webclient test
Get data...Connection status
Conn status: 0
Connection status
Conn status: 0
Connect to AP b'REDACTED' b'REDACTED'
Connection status
Conn status: 1

repeated lines removed

Connection status
Conn status: 1
Connection status
Conn status: 3
Connection status
Conn status: 3
*** Get host by name
*** Get socket
Allocated socket #0
*** Socket connect mode 2
*** Open socket
Writing: b'GET /api/v2/REDACTED/feeds HTTP/1.0\r\n'
Writing: b'Host: io.adafruit.com\r\n'
Writing: b'User-Agent: Adafruit CircuitPython\r\n'
Writing: b'X-AIO-KEY'
Writing: b': '
Writing: b'REDACTED'
Writing: b'\r\n'
Writing: b'\r\n'
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available

repeated lines removed

ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 8308 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 4308 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 308 bytes available
Reading 308 bytes from ESP socket with status 4
ESPSocket: 0 bytes available
Failed to get data, retrying
 syntax error in JSON
Reset ESP32
```

jerryneedell, 2019-02-20:
I realize the response was big - 8308 bytes -- is there a limit?

jerryneedell, 2019-02-20:
hmm -- it does look like a byte limit -- I changed "feeds" to "groups" to create a smaller response and it works fine.

If there is a limit, should there be a graceful error?

```
SPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 8308 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 4308 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 308 bytes available
Reading 308 bytes from ESP socket with status 4
ESPSocket: 5030 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 1030 bytes available
Reading 1030 bytes from ESP socket with status 4
ESPSocket: 0 bytes available
[{'updated_at': '2019-01-28T17:51:10.944124Z', 'key': 'default', 'username': 'jerryn', 'id': 110425, 'owner': {'id': 20772, 'username': 'jerryn'}, 'name': 'Default', 'description': 'This is your default group. It contains all feeds not otherwise grouped.', 'created_at': '2017-02-19T13:10:38.481454Z', 'feeds': [{'username': 'jerryn', 'visibility': 'private', 'last_value': '32.0544', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 955057, 'updated_at': '2019-02-20T20:57:01.369199Z', 'description': 'humidity from particle argon', 'name': 'argon_humidity', 'unit_symbol': None, 'created_at': '2018-12-29T21:36:24.436264Z', 'history': True, 'user_id': 20772, 'key': 'argon-humidity'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '65.0604', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 955056, 'updated_at': '2019-02-20T20:56:40.7848Z', 'description': 'temperature from particle Argon', 'name': 'argon_temperature', 'unit_symbol': None, 'created_at': '2018-12-29T21:35:55.513057Z', 'history': True, 'user_id': 20772, 'key': 'argon-temperature'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '22.299999237060547', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 648873, 'updated_at': '2019-02-20T20:53:21.5599Z', 'description': 'DHT22 in basement', 'name': 'basementhumidity', 'unit_symbol': None, 'created_at': '2017-02-19T21:19:54.680683Z', 'history': True, 'user_id': 20772, 'key': 'basementhumidity'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '64.76000137329102', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 648872, 'updated_at': '2019-02-20T20:53:20.679399Z', 'description': 'DHT22 in basement', 'name': 'basementtemperature', 'unit_symbol': None, 'created_at': '2017-02-19T21:19:33.918691Z', 'history': True, 'user_id': 20772, 'key': 'basementtemperature'}, {'username': 'jerryn', 'visibility': 'Private', 'last_value': '224', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 841736, 'updated_at': '2018-12-26T15:42:26.275899Z', 'description': None, 'name': 'Counter', 'unit_symbol': None, 'created_at': '2018-07-16T19:38:14.603073Z', 'history': True, 'user_id': 20772, 'key': 'counter'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '1', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 748434, 'updated_at': '2019-02-16T15:40:14.611664Z', 'description': '', 'name': 'Digital', 'unit_symbol': None, 'created_at': '2018-01-07T13:22:33.898769Z', 'history': True, 'user_id': 20772, 'key': 'digital'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '449', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 701136, 'updated_at': '2019-02-20T20:56:47.244199Z', 'description': None, 'name': 'eco2', 'unit_symbol': None, 'created_at': '2017-08-12T20:54:38.607322Z', 'history': True, 'user_id': 20772, 'key': 'eco2'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '117936', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 817882, 'updated_at': '2019-01-29T03:48:10.008293Z', 'description': None, 'name': 'freeheep', 'unit_symbol': None, 'created_at': '2018-06-07T10:00:54.103447Z', 'history': True, 'user_id': 20772, 'key': 'freeheep'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': 'OFF', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 830701, 'updated_at': '2018-11-20T00:19:58.919399Z', 'description': 'button to tun on/off hose', 'name': 'hose', 'unit_symbol': None, 'created_at': '2018-07-01T14:48:38.62212Z', 'history': True, 'user_id': 20772, 'key': 'hose'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '34.299999', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 717817, 'updated_at': '2018-12-08T15:22:02.239799Z', 'description': None, 'name': 'humidity', 'unit_symbol': None, 'created_at': '2017-10-08T19:00:56.081706Z', 'history': True, 'user_id': 20772, 'key': 'humidity'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '34.47644338629184', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 648834, 'updated_at': '2019-01-01T21:10:30.246999Z', 'description': 'humidity from pi3-1', 'name': 'pihumidity', 'unit_symbol': None, 'created_at': '2017-02-19T17:35:12.413564Z', 'history': True, 'user_id': 20772, 'key': 'pihumidity'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '77.52872510872055', 'enabled': True, 'owner': {'i [... cut ...] user_id': 20772, 'key': 'hue-controller.hue-motion'}, {'username': 'jerryn', 'visibility': 'private', 'last_value': '267', 'enabled': True, 'owner': {'id': 20772, 'username': 'jerryn'}, 'unit_type': None, 'id': 794031, 'updated_at': '2018-07-16T16:03:48.3564Z', 'description': None, 'name': 'hue-photocell', 'unit_symbol': None, 'created_at': '2018-04-17T21:50:51.699061Z', 'history': True, 'user_id': 20772, 'key': 'hue-controller.hue-photocell'}], 'user_id': 20772, 'visibility': 'private'}]
OK
```

makermelissa, 2019-02-20:
Good question. I didn’t even realize there was a limit before you found this. I think ideally it should attempt to grab it in chunks and assemble it if that’s possible.

jerryneedell, 2019-02-20:
i'm not sure there is a limit, but it looked suspicious.

ladyada, 2019-02-20:
the socket/requests code should do it all for you - in chunks o' 4K - we've downloaded 250KB bitmaps so that works for sure :)

jerryneedell, 2019-02-20:
so any idea what failed?

ladyada, 2019-02-20:
sorry no :/ i haven't run this demo yet - perhaps in `requests` or `esp32spi` have it dump the socket buffer?

jerryneedell, 2019-02-20:
OK - no problem -- I misunderstood :-)  Will keep looking into it.

ladyada, 2019-02-20:
np - its a mystery - and maybe i have a bug too...not sure :D also json parser may not be so great ;)

makermelissa, 2019-02-20:
I can take a closer look this evening. I don’t have my computer with me at the moment.

makermelissa, 2019-02-21:
Ah yeah, ok. I think @ladyada is correct about the JSON parser. This looks like the error message is `syntax error in JSON`, which means that it probably wan't valid JSON (this can be verified by running it through a JSON parser online). Some JSON parsers can still parse imperfect JSON, but I'm guessing ujson (the library used) is meant to be small so doesn't have really any margin of error. The thing is, there's not much we can really do in this library. Perhaps write out own parser that handles imperfect json in CP? If you want to take a look at the source of ujson, you can see it at https://github.com/adafruit/circuitpython/blob/master/extmod/modujson.c

makermelissa, 2019-02-21:
As a follow-up after looking at the code, it doesn't appear to handle single quotes, or capitalized True, False, Null, etc.

makermelissa, 2019-02-21:
Just noticed this in the comments which I formatted a little better. 

"The JSON specification is at http://www.ietf.org/rfc/rfc4627.txt
The parser here will parse any valid JSON and return the correct corresponding Python object.  It allows through a superset of JSON, since it treats commas and colons as "whitespace", and doesn't care if brackets/braces are correctly paired.  It will raise a ValueError if the input is outside its specs."

So yeah, it probably was outside its specs.

jerryneedell, 2019-02-21:
FYI - an update -- I have been running the "cheerlights" demo for a long time (>24 hours) but I notice that it had stopped updating so I connected to the REPL and say that it was connecting but reporting JSON errors!  I had not seen these with this demo before. However, rebooting the system made the problem go away - it connected and is updating normally again.  That makes no sense to me...
I hope to have some time to dig into this this weekend.  Something fishy is going on....

xelll, 2019-03-29:
> the socket/requests code should do it all for you - in chunks o' 4K - we've downloaded 250KB bitmaps so that works for sure :)

did you use this esp32 firmware？
https://cdn-learn.adafruit.com/assets/assets/000/071/843/original/NINA_W102_Feb_17_2018.bin?1550892535

ladyada, 2019-03-29:
yep thats what we use

xelll, 2019-03-30:
> yep thats what we use

ok, did you test how fast is it? i test it about 160KB/s ，but not very stable.

ladyada, 2019-03-30:
yes its not stable compared to a USB wifi adapter, its meant for simple microcontrollers!

xelll, 2019-03-30:
> yes its not stable compared to a USB wifi adapter, its meant for simple microcontrollers!

ok， how fast can it work stable.

ladyada, 2019-04-01:
there's no speed relation, it is a completely different system of using the internet. really, we recommend using a USB wifi adapter if you are on linux

xelll, 2019-04-01:
> there's no speed relation, it is a completely different system of using the internet. really, we recommend using a USB wifi adapter if you are on linux

ok...

ladyada, 2019-05-28:
@jerryneedell is this resolvedish?

jerryneedell, 2019-05-28:
I think so -- At least I have not seen it in a long time -- Closing...

jerryneedell, 2019-05-28:
oops -- closed too soon -- this issue is still present

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.1-5-g63b253c33 on 2019-05-24; Adafruit PyPortal with samd51j20
>>> import aio_test
ESP32 SPI webclient test
Get data...Connection status
Conn status: 0
Connection status
Conn status: 0
Connect to AP b'Needell Airport' b'REDACTED'
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 3
Connection status
Conn status: 3
*** Get host by name
*** Get socket
Allocated socket #0
*** Socket connect mode 2
*** Open socket
Writing: b'GET /api/v2/jerryn/feeds HTTP/1.0\r\n'
Writing: b'Host: io.adafruit.com\r\n'
Writing: b'User-Agent: Adafruit CircuitPython\r\n'
Writing: b'X-AIO-KEY'
Writing: b': '
Writing: b'REDACTED'
Writing: b'\r\n'
Writing: b'\r\n'
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 15421 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 11421 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 7421 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 3421 bytes available
Reading 3421 bytes from ESP socket with status 4
ESPSocket: 0 bytes available
Failed to get data, retrying
 syntax error in JSON
Reset ESP32
Get data...Connection status
Conn status: 0
Connection status
```
here is the test code 
```
import time
import board
import busio
from digitalio import DigitalInOut

from adafruit_esp32spi import adafruit_esp32spi
from adafruit_esp32spi import adafruit_esp32spi_wifimanager

print("ESP32 SPI webclient test")

# Get wifi details and more from a settings.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi settings are kept in secrets.py, please add them there!")
    raise

esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(esp, secrets, None)
esp._debug=True
while True:
    try:
        print("Get data...", end='')
        response = wifi.get(
            "https://io.adafruit.com/api/v2/"+secrets['aio_username']+"/feeds",
            headers={"X-AIO-KEY":secrets['aio_key']})
        print(response.json())
        response.close()
        print("OK")
    except (ValueError, RuntimeError) as e:
        print("Failed to get data, retrying\n", e)
        wifi.reset()
        continue
    response = None
    time.sleep(15)

```

jerryneedell, 2019-05-28:
retested after merging #44 and it still fails as above.

ladyada, 2019-05-28:
what do you see when you print the text out before json parsing?

jerryneedell, 2019-05-29:
I'm not sure what you are looking for... what do you want me to print out?

ladyada, 2019-05-29:
`response.text` or somethin :)

jerryneedell, 2019-05-29:
OK -- added print(response.text) before print(response.json())

```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.1-5-g63b253c33 on 2019-05-24; Adafruit PyPortal with samd51j20
>>> import aio_test
ESP32 SPI webclient test
Get data...Connection status
Conn status: 0
Connection status
Conn status: 0
Connect to AP b'Needell Airport' b'browndog'
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Failed to connect, retrying
 ('No such ssid', b'Needell Airport')
Connection status
Conn status: 1
Connect to AP b'Needell Airport' b'browndog'
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 1
Connection status
Conn status: 3
Connection status
Conn status: 3
*** Get host by name
*** Get socket
Allocated socket #0
*** Socket connect mode 2
*** Open socket
Writing: b'GET /api/v2/jerryn/feeds HTTP/1.0\r\n'
Writing: b'Host: io.adafruit.com\r\n'
Writing: b'User-Agent: Adafruit CircuitPython\r\n'
Writing: b'X-AIO-KEY'
Writing: b': '
Writing: b'94cce1f27dd2caac8da3ed48571946a147be82ee'
Writing: b'\r\n'
Writing: b'\r\n'
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
[... 12 lines cut ...]
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 324 bytes available
Reading 324 bytes from ESP socket with status 4
ESPSocket: 16384 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 12384 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 8384 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 4384 bytes available
Reading 4000 bytes from ESP socket with status 4
ESPSocket: 384 bytes available
Reading 384 bytes from ESP socket with status 4
ESPSocket: 1514 bytes available
Reading 1514 bytes from ESP socket with status 4
ESPSocket: 0 bytes available
*** Closing socket #0
[{"username":"jerryn","owner":{"id":20772,"username":"jerryn"},"id":648833,"name":"pitemperature","description":"temperature from pi3-1","license":null,"history":true,"enabled":true,"visibility":"private","unit_type":null,"unit_symbol":null,"last_value":"71.04937841796875","created_at":"2017-02-19T17:34:52Z","updated_at":"2019-05-29T10:22:16Z","status_notify":true,"status_timeout":60,"status":"online","key":"pitemperature","group":{"id":110425,"key":"default","name":"Default","user_id":20772},"groups":[{"id":110425,"key":"default","name":"Default","user_id":20772}],"feed_webhook_receivers":[],"feed_status_changes":[{"created_at":"2019-05-13T00:32:12Z","from_status":"offline","to_status":"online","email_sent":null,"email_sent_to":null},{"created_at":"2019-05-13T00:28:10Z","from_status":"online","to_status":"offline","email_sent":true,"email_sent_to":"jerryneedell@gmail.com"},{"created_at":"2019-05-12T17:31:08Z","from_status":"offline","to_status":"online","email_sent":null,"email_sent_to":null},{"created_at":"2019-05-12T15:32:10Z","from_status":"online","to_status":"offline","email_sent":true,"email_sent_to":"jerryneedell@gmail.com"},{"created_at":"2019-05-03T14:18:20Z","from_status":"offline","to_status":"online","email_sent":null,"email_sent_to":null},{"created_at":"2019-05-03T14:12:10Z","from_status":"online","to_status":"offline","email_sent":true,"email_sent_to":"jerryneedell@gmail.com"}]},{"username":"jerryn","owner":{"id":20772,"username":"jerryn"},"id":648834,"name":"pihumidity","description":"humidity from pi3-1","license":null,"history":true,"enabled":true,"visibility":"private","unit_type":null,"unit_symbol":null,"last_value":"41.863006591796875","created_at":"2017-02-19T17:35:12Z","updated_at":"2019-05-29T10:22:17Z","status_notify":true,"status_timeout":60,"status":"online","key":"pihumidity","group":{"id":110425,"key":"default","name":"Default","user_id":20772},"groups":[{"id":110425,"key":"default","name":"Default","user_id":20772}],"feed_webhook_receivers":[],"feed_status_changes":[{"created_at":"2019-05-13T00:32:12Z","from_status":"offline","to_status":"online","email_sent":null,"email_sent_to":null},{"created_at":"2019-05-13T00:28:10Z","from_status":"online","to_status":"offline","email_sent":true,"email_sent_to":"jerryneedell@gmail.com"},{"created_at":"2019-05-12T17:31:08Z","from_status":"offline","to_status":"online","email_sent":null,"email_sent_to":null},{"created_at":"2019-05-12T15:32:10Z","from_status":"online","to_status":"offline","email_sent":true,"email_sent_to":"jerryneedell@gmail.com"},{"created_at":"2019-05-03T14:18:20Z","from_status":"offline","to_status":"online","email_sent":null,"email_sent_to":null},{"created_at":"2019-05-03T14:12:10Z","from_status":"online","to_status":"offline","email_sent":true,"email_sent_to":"jerryneedell@gmail.com"}]},{"username":"jerryn","owner":{"id":20772,"username":"jerryn"},"id":648872,"name":"basementtemperature","description":"DHT22 in basement","license":null,"history":true,"enabled":true,"visibility":"private","unit_type":null,"unit_symbol":null,"last_value":"57.940039062500006","created_at":"2017-02-19T21:19:33Z","updated_at":"2019-05-29T10:22:48Z","status_notify":true,"status_timeout":60,"status [... cut ...] 19T20:12:10Z","status_notify":false,"status_timeout":4320,"status":"offline","key":"rfm69vbat","group":{"id":110425,"key":"default","name":"Default","user_id":20772},"groups":[{"id":110425,"key":"default","name":"Default","user_id":20772}],"feed_webhook_receivers":[],"feed_status_changes":[{"created_at":"2019-04-19T20:12:10Z","from_status":"online","to_status":"offline
Failed to get data, retrying
 syntax error in JSON
Reset ESP32
Get data...Connection status
Conn status: 0
Connection status
```

ladyada, 2019-05-29:
yeah looks like its incomplete - dunno why tho

kattni, 2020-05-04:
@jerryneedell Have you made any progress on this?

us3r64, 2024-01-15:
@ladyada it might be related to this: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/issues/184