adafruit/Adafruit_CircuitPython_ESP32SPI #13: adafruit_esp32spi_wifimanager: allow DotStar as status pixel?

anecdata, 2019-02-26:
Since ItsyBitsy M4 is a viable main MPU to use with the ESP32 co-processor, would be nice to be able to use the built-in DotStar as the status pixel instead of adding an external NeoPixel.

ladyada, 2019-02-26:
yeah honestly we should just default to the built in LED - want to try a PR?

anecdata, 2019-02-26:
I'm 95% n00b at [Circuit]Python and 100% n00b at git[Hub] tools & process, wouldn't trust me to modify libraries at this point. Happy to test stuff though, I'm good at breaking code ;-)

anecdata, 2019-02-26:
Thought there'd be a simple abstraction or conditional possible, but DotStar not in n HAL or core modules. The CP RGB status LED works for both neos and dots though (CPX ibeing special), so maybe some abstraction there. Or... cheerlights uses fancyled?

jerryneedell, 2019-02-26:
@ladyada Are you suggesting using the builtin LED (D13 on most boards) instead of the neopixel? It would still be nice to pass in a Pin since there are some odd cases (particle boards for instance). But are you ok with just an on/off for the LED instead of the colors the neopixel now uses or do you want to preserve the option for a neopixel. Should we have 2 optional kwargs -- one for a Neopixel, one for an LED -- if None -- ignore them -- if set -- use them.

makermelissa, 2019-02-26:
I think this is a great idea! I didn't realize the ItsyBitsy had a DotStar. I'm going to think about the best way to implement this.

ladyada, 2019-02-26:
ideally you'd pass in a NeoPixel or DotStar *object* that you'd call `fill` on - no instantiation in the library itself

makermelissa, 2019-02-26:
Yeah, that's kind of along the lines of what I was thinking. That's about how I did the DotStar/NeoPixel abstraction in the FeatherWing library.

makermelissa, 2019-02-26:
Fixed #13 via #14.

closed 2019-02-26 by makermelissa