adafruit/Adafruit_CircuitPython_ESP32SPI #22: HTTP server response headers are not cleared between fetches

anecdata, 2019-03-08:
This issue will probably have to get moved. I'm seeing some interesting behavior with web server response headers, not sure where to put the issue. Somewhere, I don't think it's here in ESP32SPI, but then must be either NINA FW or the depths of CP.

Headers not served by the current server are delivered back to the client. I see this on every HEAD or GET from virtually any site after the first site. After pulling a couple of pages from disparate sources, an accumulation of unique headers from prior server responses are presented to the client in `response.headers` along with the correct headers from the current site.

Clearing the response in the client using `response = None` does not clear the headers, the accumulated set of headers is still returned on subsequent calls.

anecdata, 2019-03-09:
Code and output in a Gist here: [https://gist.github.com/anecdata/922abc20112a442a4fed46bc1f6ad493](url)

This happens with fetches using either adafruit_esp32spi_wifimanager.py or using adafruit_esp32spi_requests.py directly.

ladyada, 2019-03-09:
hiya that gist is private? or...we cant see it?

anecdata, 2019-03-09:
Should be public, I can open it in a Private tab. Maybe I was editing?

ladyada, 2019-03-09:
oh you have to copy and paste - click wont work

anecdata, 2019-03-09:
Sorry, operator error, link fixed.

makermelissa, 2019-03-09:
Thanks @anecdata. I just tried it on my PyPortal and am seeing the same results as you.

PR #23 (2019-03-09): Clear the headers in the init
To fix #22, I moved the clearing the headers into the init function rather than the declaration and that seems to solve it.
closed 2019-03-09 by makermelissa