adafruit/Adafruit_CircuitPython_ESP32SPI #24: Adding HTTP request headers doesn't work (adafruit_esp32spi_requests.py)

anecdata, 2019-03-09:
There is a facility for any HTTP request (HEAD, GET, POST, etc.) to have user-specified HTTP headers added. For example, 'Http-X-MyCustomHeader: BlahBlahBlah'.
```
def request(method, url, data=None, json=None, headers=None, stream=False):
...'headers' is optional HTTP headers sent along.
```
Current version pieces together the header components to write the header to the socket based on user-specified dict containing one or more key-value pairs describing the header name and value, but there's a runtime error when trying to process the dictionary I believe due to unencoded string elements of the headers.

To fix, I _think_ it's just a matter of changing lines 176 & 178 in adafruit_esp32spi_requests.py to encode the strings into (default ASCII) bytestrings, changing from:
```
        for k in headers:
            sock.write(k)
            sock.write(b": ")
            sock.write(headers[k])
            sock.write(b"\r\n")
```
to:
```
        for k in headers:
            sock.write(k.encode())
            sock.write(b": ")
            sock.write(headers[k].encode())
            sock.write(b"\r\n")
```
I tested it in a basic scenario, and the program runs, and I receive the custom request headers on my server. But I'd really appreciate more seasoned Python eyes looking at this.

makermelissa, 2019-03-09:
Hi, do you want to submit a PR and then some more “seasoned” eyes could take a look at that during the review? 🙂

anecdata, 2019-03-09:
Sure, I'll give it a shot. There's a first time for everything. I'll do some reading on the process and see what I need to set up to do that.

makermelissa, 2019-03-09:
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

brentru, 2019-03-11:
Fixed in release 1.3.4! Closing...

closed 2019-03-11 by brentru