adafruit/Adafruit_CircuitPython_ESP32SPI #28: requests fails sending large header values or paths over socket

martymcguire, 2019-03-13:
I'm trying to make an HTTP request with a ~356 byte access token, and when making the request via `requests.get(...)` it fails with:

```
Failed to send 363 bytes (sent 107)
```
With the ESP's `_debug` turned on, I see something like this (redacted for token privacy):

```
...
*** Get host by name
*** Get socket
Allocated socket #0
*** Socket connect mode 2
*** Open socket
Writing: b'GET /microsub/1?action=timeline&channel=duQSjrTyDhvn2BQvrs7jLkV5v4Q4ZYw3 HTTP/1.0\r\n'
Writing: b'Host: aperture.maktro.net\r\n'
Writing: b'User-Agent: Adafruit CircuitPython\r\n'
Writing: b'authorization'
Writing: b': '
Writing: b'Bearer XXXX...FOR_356_BYTES...XXX'
Failed to send 363 bytes (sent 107)
...
```
The server I'm talking to allows me to pass the token in the query string, but I end up with a very similar error (failing on the `GET /microsub/1?...` line).

If I truncate the value of the access token so it's less than ~100 bytes, the request goes through, but the server of course fails it because it's not a valid access token.

It looks like I'm running into an issue with this code: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/adafruit_esp32spi/adafruit_esp32spi.py#L535

Is it possible to have the requests lib chunk these sends over the socket?

Thanks!!

ladyada, 2019-03-13:
yep - i think we just haven't sent large chunks of data before, we split into 64 bytes! or you can try to do it at the socket-send level if you want to try :)

kattni, 2020-04-29:
This was resolved in #29 .

closed 2020-04-29 by kattni