adafruit/Adafruit_CircuitPython_ESP32SPI #33: Spaces in SSID results in couldn't connect and 'No such ssid'

warthog9, 2019-04-04:
Running the sample code from https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/examples/esp32spi_simpletest.py

On a PyPortal and using:
```
SSID: 'MY SSID NAME'
Password: 'My Super Secure WiFi Password'
```
Noting the SSID has a space in it.

https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/c2e0fd2b094d187f4aa52f7180c7a11b652250fc/examples/esp32spi_simpletest.py#L33 

results in:
```
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.2.2\x00')
MAC addr: ['0xAA', '0xBB', '0xCC', '0xDD', '0x00', '0x11']
        MY SSID NAME            RSSI: -56
        NeighborFi              RSSI: -57
        OtherNeighborFi           RSSI: -57
        BillWitheScienceFi         RSSI: -57
        MY SSID NAME        RSSI: -57
        Linksys        RSSI: -57
        Open Free Wifi        RSSI: -57
`could not connect to AP, retrying:  ('No such ssid', b'MY SSID NAME')`
```
Also noting that, yes, I'm running multiple access points with `MY SSID NAME`, they are working fine particularly with other ESP32 and ESP8266 devices that don't happen to run CircuitPython.

Switching to an SSID without spaces works fine however.  Tested against several (4) networks that work that have spaces in them, and against one that does not.  Failed, identically, on the 4 networks with spaces, and worked fine with the network that did not.

ladyada, 2019-04-04:
ha - do you have the ability to try it on arduino IDE? just curous if its a python thing or an ESP32 thing.

warthog9, 2019-04-04:
Ok, this might be a false alarm and indicative of a bug not in this code.  I thought I had a reliable test case, but in poking at it some more it started failing in different ways.  Might be an issue in the code on my access points, as disabling them and re-enabling them to try and isolate what was going on further changed the pathology, and giving one of them a swift kick in the reboot button seems to have cleared up the problem.

closed 2019-04-04 by warthog9

rudypospisil, 2021-08-19:
I had this issue and for me, it was the spaces in my SSID. I removed them and now all is well.