adafruit/Adafruit_CircuitPython_ESP32SPI #35: Companion firmware binaries

rdagger, 2019-04-07:
Are there pre-built binaries for the companion firmware?

docmollo, 2019-04-07:
The official firmware is available from the Learn guide (https://learn.adafruit.com/adding-a-wifi-co-processor-to-circuitpython-esp8266-esp32/firmware-files#esp32-only-spi-firmware-3-8). If you're looking for a pre-compiled binary with WPA2 Enterprise support, that hasn't been released yet, as the support just got merged.

ladyada, 2019-04-07:
@doc dya wanna attach a file here and then i can put it in the learn guide

docmollo, 2019-04-07:
Sure! Here ya go.

[NINA_W102-1.3.0.zip](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/files/3051756/NINA_W102-1.3.0.zip)

ladyada, 2019-04-07:
@rdagger please try it!

rdagger, 2019-04-07:
@ladyada Thanks, it works great!

docmollo, 2019-04-07:
Thanks @rdagger! You might be the first person to try out the new code...certainly the first to try out my binary firmware file. Glad to hear it all works for you! :blush:

ladyada, 2019-04-07:
https://learn.adafruit.com/adding-a-wifi-co-processor-to-circuitpython-esp8266-esp32/firmware-files

closed 2019-04-07 by ladyada