adafruit/Adafruit_CircuitPython_ESP32SPI #36: Expose additional NINA FW command handlers (AKA Release the Kraken)

anecdata, 2019-04-11:
Expose additional command handlers from CommandHandler.cpp from the adafruit/nina-fw, especially setPinMode, setDigitalWrite, and setAnalogWrite.

There are also other potentially interesting command handlers, but I'm not sure which are fully implemented even in the C code; I don't seem to get much meaningful returned from getTime or getTemperature for example (I could easily be doing something wrong).

TIn Arduino, the Airlift breakout can control its on-board red, green, and blue LEDs via the adafruit/WiFiNINA library. But for CircuitPython on a separate M4 processor, the ESP32SPI library needs to expose the GPIO command handlers to be able to control the GPIO pins.

Potentially, _many_ ESP32 pins could be made available to CircuitPython, depending on form factor:

• ESP 32 Airlift breakout GPIO/A pins: 25[Red LED], 26[Green LED], 27[Blue LED]

• ESP32 Huzzah32 breakout GPIO/A pins: 15,#2,#0,#4,16,17,#5,18,19,21,22,23,13,12,14,27,26,25,33,32,35,34,39,36

• ESP32 Huzzah32 Feather GPIO/A pins: A0/26,A1/25,A2/34,A3/39,A4/36,A5/4,21,13[LED],12[PULLDOWN],27,33,15,32,14

• ESP32 Airlift Feather GPIO/A pins (speculation based on #LEEK): A0/26, A1/25, A2/34, A3/39

[ P.S. I spent several hours trying to do this myself, but between the python code, SPI protocol, and layers of C code, it's beyond my current level. I'll keep digging though. I'm sure someone more skilled in the arts would look at it and say "oh, of course you'd just..." ;-) ]

cpforbes, 2019-04-23:
I just submitted PR #38 that exposes the low level GPIO commands already in the nina-fw. Nothing specific to a particular ESP32 board.

ladyada, 2019-04-23:
nice!
anyone wanna test it out? @brentru should be able to twiddle the RGB LEDs on the AirLift breakout

brentru, 2019-04-23:
Yup, I’ll test and review tomorrow.

anecdata, 2019-04-23:
Airlift Breakout RGB LEDs are on! [image]

ladyada, 2019-04-23:
:D

anecdata, 2019-04-23:
Pin #13 RED LED on Feather ESP32 is on! 
[image]

brentru, 2019-04-25:
Merged into master and released!

closed 2019-04-25 by brentru