adafruit/Adafruit_CircuitPython_ESP32SPI #37: WiFiManager debug not working

rdagger, 2019-04-18:
An error is thrown upon connection if you set debug to true:
```
wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(esp, secrets, None)
wifi.debug = True
```
> File "adafruit_esp32spi/adafruit_esp32spi_wifimanager.py", line 108, in get
> File "adafruit_esp32spi/adafruit_esp32spi_wifimanager.py", line 72, in connect
> AttributeError: 'module' object has no attribute 'WL_IDLE_STATUS'

Also the status_pixel parameter is supposed to be optional, but it's not.

rdagger, 2019-04-19:
Disregard regarding debugging.  I figured out that you can't set the debug from the WifiManager.  Instead it has to be set when you initialize adafruit_esp32spi.
```
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset, debug=True)
wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(esp, secrets, None)
```
But the status_pixel parameter is supposed to be optional.

brentru, 2019-04-25:
The latest release fixes the debug parameter. Do you still see this issue when using it?

rdagger, 2019-04-26:
Yes, it's working.  Thanks.

PR #40 (2019-05-05): Fix debug in WiFiManager and socket leak in request().
Also adds context manager methods so the response can be used in a with to automatically close the socket.

Fixes #37
closed 2019-04-26 by rdagger