adafruit/Adafruit_CircuitPython_ESP32SPI #41: Using Adafruit.io weather API, responses often truncated

MisterC925, 2019-04-26:
I've tried adapting the PyPortal weather example to use the Dark Sky data from Adafruit.io

I'm still using the PyPortal library from adafruit_pyportal as in the original code. 

When pyportal.fetch() is called, at least 50% of the time the JSON is truncated, resulting in an exception for json.loads.

I've noticed that successful fetches gather around 6k of data and the failing ones are all truncated at 3214 bytes (this is the length of the results of the pyportal.fetch, so I don't believe it has headers in it) 

Please Advise.

Thanks!

lemonkey, 2019-07-18:
I see this for images as well.

ymauray, 2019-07-27:
This is really anoying, especially when getting forecast data from OpenWeatherMap : 3 hours periods = 8 periods per day, * 5 days for a nice forecast = 40 periods, and there is a ton of data per period.

Is the data truncated because of available memory or is it just arbitrary ?

ladyada, 2019-07-27:
try turning on debug mode and looking at where the failure is happening, that will help!

ymauray, 2019-07-27:
I realize my test case was using a swiss weather service, not OpenWeatherMap. 
Here's my test case :
I changed  DATA_SOURCE to reference the Swiss service 
```
DATA_SOURCE = "http://www.prevision-meteo.ch/services/json/geneve"
```
This returns a huge json file with a ton of info. 
Once debug is turned on on PyPortal, the json is printed and I can see that it is truncated :
```
Response is {"city_info":{"name":"Gen\u00e8ve","country":"Suisse",[data truncated for brievety],"WNDDIRCARD10":"O","ISSNOW":0,"HCDC":"100.00","MCDC":"5.20","LCDC":"0.00","HGT0C":
Trace (appels les plus récents en dernier):
  Fichier "main.py", ligne 66, dans <module>
  Fichier "main.py", ligne 62, dans <module>
  Fichier "openweather_graphics.py", ligne 65, dans display_weather
ValueError: erreur de syntaxe JSON
```

Line 64 and 65 of openweather_graphics.py are
```
    def display_weather(self, weather):
        weather = json.loads(weather)
```
which is called from main program by 
```
            value = pyportal.fetch()
            print("Response is", value)
            gfx.display_weather(value)
```

So maybe this file is just too big to fit in the PyPortal memory ?
If so, can there be a specific error message ? if the payload size differs from what is specified in the response's headers, maybe ?

ladyada, 2019-07-28:
hard to tell but yeah thats a huge JSON - see if you can get one day at a time

kattni, 2020-05-04:
@MisterC925 Are you still experiencing this issue?

closed 2020-06-30 by dherrada