adafruit/Adafruit_CircuitPython_ESP32SPI #46: Analog read

nezra, 2019-05-23:
given the huzzah32 with it's battery & voltage divider attached directly to one of the analog pins;  and the release the kraken patch, #34?, would it be possible to add an analog read function? and maybe a digital read?

I'd like to use the built in battery functionality of the huzzah32 to power a portion of a project; and possibly use it for supplementary IO; but i'd want to monitor the voltage. the added voltage regulator & power storage would be able to provide  an easier way than making a voltage regulator tacked a feather & it's battery pin.

I don't pretend to understand the nina-fw and the command handler language; but it reminds me alot of arduino. after reading through it, would something like:

int setAnalogRead(const uint8_t command[], uint8_t response[])
{
  uint8_t pin = command[4];
  unit8_result = 255

  result=analogRead(pin);

  response[2] = 1; // number of parameters
  response[3] =  sizeof(result); // parameter 1 length
  memcpy(&response[4], result, sizeof(result));

  return 9;
}

be sufficient?

ladyada, 2019-05-23:
you could but yeah you'd have to add the matching capability to nina-fw as well - we don't have plans to add it but feel free to!

anecdata, 2019-05-23:
Your issue on NINA could be combined with #19[https://github.com/arduino/nina-fw/issues/19](url)

That's the place to start. Once (if) it's implemented in the NINA firmware, it's straightforward to expose it from the ESP32SPI library.

anecdata, 2019-10-17:
[PR](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/80) for #76 has been submitted.

brentru, 2019-10-23:
Close via https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/80

closed 2019-10-23 by brentru