adafruit/Adafruit_CircuitPython_ESP32SPI #48: Runtime Errors in WiFiManager when posting in a loop

turbinenreiter, 2019-05-29:
The following (simplified) code snippet:

```
while True:
    package = {
        some dictionary
    }
    try:
        wifi.post("http://159.69.220.75:8000/pub", json=package)
        print("+", end="")
    except (ValueError, RuntimeError):
        print("-", end="")
        wifi.reset()
        wifi.connect()
        print("reconnected")
    gc.collect()
    time.sleep(1)
```

leads to this output:

```
++++++++++-reconnected
++++++++++-reconnected
++++-reconnected
+++++++++-reconnected
++++++++++-reconnected
++++++++++-reconnected
++++++++++-reconnected
++++++++++-reconnected
++++++++++-reconnected
++++++++++-reconnected
```

So you can see that it pretty reliably does 10 requests, fails at the 11th and then resets.

Although it is easy to make your code robust against this issue, the consistency of the number of requests you can do makes me think that there might be a fixable underlying issue worth investigating.

kattni, 2020-05-04:
@turbinenreiter Are you still experiencing this issue?

turbinenreiter, 2020-05-05:
I haven't tried since, and honestly I can't even remember in which of my projects I had this issue :D

robmarkcole, 2020-06-26:
I am also experiencing this when using with the minimqtt lib in a `while True` loop on a pyportal. At a random time [my script ](https://github.com/robmarkcole/Useful-python/blob/master/Circuitpython/pyportal/code.py)will fail with `ESP32 not responding`. I understand this is intended behaviour, so what is required is a strategy to deal with it. Perhaps placing the entire script within a try, excepet, rerun loop?

brentru, 2020-06-26:
@robmarkcole 
> so what is required is a strategy to deal with it. Perhaps placing the entire script within a try, excepet, rerun loop?
Yes, we do something similar in the Adafruit IO post example here: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/examples/esp32spi_aio_post.py#L69

dherrada, 2020-06-30:
Appears to be resolved. @robmarkcole Let me know if you're still having issues with this and I can reopen

closed 2020-06-30 by dherrada