adafruit/Adafruit_CircuitPython_ESP32SPI #49: Variable Request Timeouts

anecdata, 2019-05-29:
adafruit_esp32spi_requests.py `request()` sets 'socks.settimeout()' to 1 second. This is fine in many cases, but due to server, traffic, and various other issues, it may not be enough to reliably fetch data. Retries can partially resolve this, but more user control over timeouts would give users the ability to make their own trade-offs between speed and reliability.

I'd propose initially making the timeout within `request()` a keyword argument, and rippling it up to WiFiManager and device-specific classes like PyPortal.

Other than ping ttl and hard-coded delays and timeouts deep in ESP_SPIcontrol, everything in ESP32SPI appears to rely on the 1-second timeout set in the request (notably in socket read and readline). Other than ping and wifi scans, NINA firmware doesn't seem to introduce any substantial timeouts that I could find. I'm not sure what the ESP core does. I haven't tested to determine if there's a max name lookup time within the system (DNS servers will eventually time out and provide an error response).

Ideally (long-term), perhaps users could have separate control over timeouts for name lookup, connections, and requests. Browsers have many-second timeouts for connections and requests. cURL also has separate connection and overall timeouts, with relatively high defaults (300 on connection, ∞ on the overall request). We don't need to be concerned about persistent connections for now since ESP32SPI uses HTTP/1.0 and there is no keep-alive.

anecdata, 2019-05-31:
I can do a PR. I would like someone who is more pythonically experienced and has a good sense of the architecture to confirm whether this is a reasonable approach, or suggest an alternative.

I'll run this change on a bunch of devices all weekend. Experience so far is much better reliability with programmatic control of timeout.

Another reference point: @kenneth-reitz's @requests library supports connection timeouts (default=∞)and read timeouts, (or a single value to be used value for both), and docs note that:
> It’s a good practice to set connect timeouts to slightly larger than a multiple of 3, which is the default TCP packet retransmission window.
[https://2.python-requests.org/en/master/user/advanced/#timeouts](https://2.python-requests.org/en/master/user/advanced/#timeouts)

brentru, 2019-05-31:
@anecdata when you submit your PR, tag the CircuitPython Librarians group for a code review.

makermelissa, 2019-05-31:
It sounds like a reasonable approach to me. :)

anecdata, 2019-05-31:
OK, thanks. I'll test for a while longer, then submit.

closed 2019-06-07 by anecdata