adafruit/Adafruit_CircuitPython_ESP32SPI #63: Connection Sleep or Timeout

calittle, 2019-08-13:
This is an enhancement request to provide for dropping a connection and shutting off the radio after a specified time period of no use or after a fixed time period. Upon request e.g. post(), WifiManager will turn on the radio and reestablish the connection and reset the timer. This could be useful for battery-powered projects where always-on radio and connections will drain the battery quickly.

siddacious, 2019-08-13:
@brentru Does the NINA firmware have a shutdown feature?

mscosti, 2019-08-24:
I'm not sure exactly what 'low power mode' does, but NINA does support putting it into a low power mode: https://github.com/adafruit/nina-fw/blob/master/main/CommandHandler.cpp#L148-L162

ladyada, 2019-08-24:
thats as good as yr gonna get. anyone want to implement that function as a PR? :)

anecdata, 2021-01-15:
The two pull requests above summarize the current state of power control of the ESP32 NINA co-processor:
- NINA sets a nominal lower power mode by default that reduces power usage about roughly half when not communicating with the AP. This is the normal level of opereation already. There is a higher power saving mode that would require NINA and ESP32SPI changes, but also may introduce operational differences.
- A `disconnect()` function is now available in `ESP_SPIcontrol`, reducing average power usage to about half in common use cases. This function can now be used directly in user code, or could be incorporated into higher-level libraries like WiFiManager.

brentru, 2021-01-15:
> A disconnect() function is now available in ESP_SPIcontrol, reducing average power usage to about half in common use cases. This function can now be used directly in user code, or could be incorporated into higher-level libraries like WiFiManager.
Exciting!

anecdata, 2021-01-17:
Just a note that ESP32SPI NTP using `esp.get_time()` doesn't play well with staying disconnected until needed since the ESP32 needs some interval (~15 seconds I believe) after connection before it will return valid seconds.