adafruit/Adafruit_CircuitPython_ESP32SPI #72: get_time( returns (0,) for awhile

jerryneedell, 2019-09-13:
while investgating this issue
https://github.com/adafruit/Adafruit_CircuitPython_NTP/issues/2

I found that esp.get_time() sometimes returns (0,) after first connectng.
It eventually returns a good value but it can take >15 seconds

using this test code"
```
import time
import board
import busio
from digitalio import DigitalInOut
from adafruit_esp32spi import adafruit_esp32spi

# If you are using a board with pre-defined ESP32 Pins:
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

print("Connecting to AP...")
while not esp.is_connected:
    try:
        esp.connect_AP(b"Needell Airport", b"browndog")
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue

while True:
    print(esp.get_time())
    time.sleep(1)

```
you can see the problem 

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Adafruit CircuitPython 5.0.0-alpha.1-2-gfebfc7e4e on 2019-09-02; Adafruit PyPortal with samd51j20
>>> 
>>> import gettime
Connecting to AP...
could not connect to AP, retrying:  ('No such ssid', b'Needell Airport')
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(0,)
(1568372023,)
(1568372024,)
(1568372025,)
(1568372026,)
(1568372027,)
(1568372028,)
```

ladyada, 2019-09-13:
wanna add an exception raise?

jerryneedell, 2019-09-13:
Sure. Just raise ValueError? Let caller decide whether to retry?

ladyada, 2019-09-13:
yeppers

PR #73 (2019-09-15): raise error in get_time if zero returned
fixes #72 
on startup, sometimes it takes awhile (up to 15 seconds) for get_time() to return a non-zero result.
Raising the ValueError lets the user retry if desired

tested on PyPortal and Metro_m4_express with  with  Nina v 1.3.0 
test code:
```
import time
import board
import busio
from digitalio import DigitalInOut
from adafruit_esp32spi import adafruit_esp32spi

# If you are using a board with pre-defined ESP32 Pins:
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

spi = b [... cut]
closed 2019-09-15 by jerryneedell