adafruit/Adafruit_CircuitPython_ESP32SPI #83: Request to include datagram type for sockets

cs0ckn, 2019-11-24:
Is it possible to enhance the `adafruit_esp32spi_socket.socket` class to include datagram (UDP) socket type? This needed to allow connections with UDP only services like NTP.

docmollo, 2019-11-24:
I don't know if there are plans to add UDP socket support, but you mentioned NTP, so I thought I'd let you know the NINA firmware on the ESP32 does NTP and that time is available via this library's get_time() in case that is useful.

brentru, 2019-11-25:
@cs0ckn Yep - but you will need to do some hacking to expose it. As mentioned by @docmollo, there's a NTP method in ESP32SPI which uses the ESP-IDF's NTP API directly (nice helper library here: https://github.com/adafruit/Adafruit_CircuitPython_NTP) 

In `adafruit_esp32spi`, you can specify the `conn_mode` as `UDP_MODE` (https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/adafruit_esp32spi/adafruit_esp32spi.py#L616). 

Feel free to submit a PR!

anecdata, 2021-09-15:
There was a PR to add socket-level (datagram) UDP <https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/127>, example here <https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/main/examples/esp32spi_udp_client.py>.

closed 2021-09-15 by anecdata