adafruit/Adafruit_CircuitPython_ESP32SPI #84: set_analog_read value returned

nezra, 2019-12-04:
issue or not, looking for some clarification as to why.

when  set_analog_read is returned, the result is passed but first multiplied by 16. 

This means the resultant value has to be divided by 16 to get an accurate value. is there a reason for this? i used a huzzah32, with a voltage divider of :2 with 2 100k ohm resisters tied to a lipo battery and to an analog pin.

the resultant formula should be

`battery=esp.set_analog_read()/4096*7.445`

however, since the ESP32SPI passes back the value multiplied by 16 it didn't work and took me forever to figure out why. the docs didn't make it quite clear that to get an accurate number you needed to divide it by 16. i know it mentions the range being 0-65536, but as there are no examples, and any examples in arduino and the traditional way of doing analog reads aren't pre-ranged, it makes it very confusing to try and back track/read through the libraries to figure out whats wrong.

brentru, 2019-12-04:
cc @anecdata who worked on the analog read implementation

anecdata, 2019-12-04:
Yes, ESP32 natively uses a 12-bit value by default (but it can be changed to be as small as 9-bit), but for ESP32SPI a 16-bit value was chosen to be consistent with CircuitPython 'analogio'. A bit arbitrary perhaps, especially since ESP32SPI analogWrite uses a 0.0-1.0 value. I don't have a strong position on this. Certainly we could find a way to document the behavior.

anecdata, 2019-12-04:
Should also note that attenuation is a kwarg, and this affects how to interpret the voltage from the returned value.

nezra, 2019-12-04:
The logic is sound, and you've addressed my question in a pleasant and descriptive manner. i would simply request it be documented slightly more in depth, and maybe a simple example for both read and write since the values are inherently different between the two functions(maybe with one of those amazing 'how to do it' adafruit blog tutorials.

I believe the consistency with the analogio makes much more sense in the long run. i'll close the comment as addressed, as its not really an issue but an intentional and logical method.

closed 2019-12-04 by nezra