adafruit/Adafruit_CircuitPython_ESP32SPI #94: adafruit_esp32spi_socket.py:readline hardcoded end of line marker

graemorrison, 2020-03-25:
readline in adafruit_esp32spi_socket uses a hardcoded end of line marker as '\r\n', this obviously does not work for all systems, it should be made definable.

I have updated the routine to do this and tested it out on a pyportal but do not have permissions to push a new branch back to the repo, please see the attached updated code (used a .txt suffix as .py is not allowed)

[adafruit_esp32spi_socket.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/files/4380597/adafruit_esp32spi_socket.py.txt)

regards,
Grae Morrison

FoamyGuy, 2020-03-25:
@graemorrison if you are up to it can you go ahead and make a Pull Request with these changes? There is a guide here on how to do this for circuitpython projects: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github. You'll make a fork of the project and then make your changes within a branch of your fork. Then come back to this repo and create the Pull Request selecting your fork/branch to get merged into the main one. If not no worries I can try to get it created later tonight.

graemorrison, 2020-03-25:
Thanks for the pointer @FoamyGuy - I'm more than happy to go down that road, will get on with it and let you know,

tekktrik, 2022-01-14:
Resolved by PR #150!

FoamyGuy, 2022-01-17:
closed by #150

closed 2022-01-17 by FoamyGuy