adafruit/Adafruit_CircuitPython_ESP32SPI #100: Add a reconnect method to the WiFi manager

jimbobbennett, 2020-06-02:
There is currently no single method way to reconnect the WiFi manager if the connection is lost. Currently it's a two step process:

```python
wifi.reset()
wifi.connect()
```

It would be nice to have a single call to do this - `wifi.reconnect()`

jimbobbennett, 2020-06-02:
Update to this - the WiFi manager get/post/put methods all initiate a connection if not connected, so when using these methods `reset` is enough to recycle the connection.

The problem comes when using the socket directly, for example when using the MQTT libraries or the ESP object directly. In this case `connect` needs to be called explicitly.

This leads to three options that I can see:

1. Do nothing and rely on any code that uses the WiFi manager and the socket to reconnect manually by calling `reset` then `connect`
1. Add a new `reconnect` method
1. Make `reset` call `connect`

Thoughts?

brentru, 2020-06-02:
> 3. Make reset call connect

I do not like this option, `reset` resets the physical hardware. A `reconnect` method could perform a hardware reset, check the module status, then reconnect the socket. MiniMQTT is designed to resubscribe to existing topics.

> The problem comes when using the socket directly, for example when using the MQTT libraries or the ESP object directly. In this case connect needs to be called explicitly.

Which examples have this issue?

jimbobbennett, 2020-06-02:
An example would be the Azure IoT library - this uses the socket and network interface which are passed to miniMQTT. If code uses the WiFi manager to connect, then gets a connection issue and needs to reconnect that involves resetting the WiFi manager and reconnecting.

Or am I overthinking this, and should add a connection check and auto reconnect to the IoT library when reconnecting?

tannewt, 2020-06-02:
I think this is the crux of the API design for CircuitPython networking libraries. I don't know what the right answer is.

It feels like the protocol level (HTTP, MQTT) will need to know what to do on reconnect in order to get back into it's current state. Python's socket doesn't have a `reconnect` but it does have `connect`. Perhaps a protocol layer should be able to call connect again and it can internally do any reset of external hardware necessary.

brentru, 2020-06-19:
@tannewt 
> Perhaps a protocol layer should be able to call connect again and it can internally do any reset of external hardware necessary.

I like this. The socket which already has access to the interface, `wiznet`/`esp`, would handle network reconnection within `connect`. So if a `recv` times out, the protocol (`minimqtt`/`requests`) could call `connect` to reset and reconnect the connection?

tannewt, 2020-06-19:
Yeah, by basically getting a new socket.