adafruit/Adafruit_CircuitPython_ESP32SPI #109: parse_header Error trying to import adafruit_esp32spi.adafruit_esp32spi_wsgiserver

team3572, 2020-10-02:
I'm new to the entire CircuitPython library.
Was following an online example to create a HTTP server and this line in the example is throwing an error...

`import adafruit_esp32spi.adafruit_esp32spi_wsgiserver as server`

Error is..
```
Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py", line 34, in <module>
ImportError: cannot import name parse_headers
```
I do have the adafruit_requests library installed.

But maybe I have something incorrect in my installation.
I do have other things working using CircuitPython so in general I am running code fine.

tannewt, 2020-10-02:
Oops! I think I broke this. `parse_headers` is not actually part of the requests API. We'll need to pull the implementation into this library. The code is here: https://github.com/adafruit/Adafruit_CircuitPython_Requests/blob/master/adafruit_requests.py#L268

Do you want to try that? You can get guidance on [our Discord](https://adafru.it/discord)

Anton-2, 2020-11-12:
Fixed in #115 . With this PR I was able to make a WSGI app on a MatrixPortal allowing uploading of pictures...

tekktrik, 2021-12-29:
Looking through old issues, it looks like this may be fixed?

anecdata, 2024-05-17:
Closing. The ESP32SPI WSGI server was deprecated in #167 and #181, in favor of https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer

closed 2024-05-17 by anecdata