adafruit/Adafruit_CircuitPython_ESP32SPI #113: Unstable connectivity with Matrix Portal and most of the internet demos

diomark, 2020-11-03:
Having issues w/ the Matrix Portal and the internet demos. The only internet demo that works is the Quote board matrix display.

Using adabox016 components
adafruit-circuitpython-matrixportal_m4-en_US-6.0.0-rc.1.uf2
adafruit-circuitpython-bundle-6.x-mpy-20201101.zip

It takes a very long time  (~30seconds-1minute) to get a wifi connection and grab quotes with the Quote board matrix display. 

Serial shows these types of errors while connecting -

Socket missing recv_into. Using more memory to be compatible
An error occurred, retrying! 1 - Failed to send 3 bytes (sent 0)

Again, the quotes demo is the only one that worked. Getting errors around adafruit_requests?) in both the weather clock and the demo that pulls examples from adafruit..

ladyada, 2020-11-03:
can you get the very latest bundle? a few fixes have gone in recently

diomark, 2020-11-03:
I grabbed the latest nightly bundle last night with same results. Is there anything new since? I'll check.

diomark, 2020-11-03:
Just checked - already running latest bundle from https://circuitpython.org/libraries (adafruit-circuitpython-bundle-6.x-mpy-20201101.zip)  - also I've tried both adafruit-circuitpython-matrixportal_m4-en_US-6.0.0-rc.1.uf2 as well as adafruit-circuitpython-matrixportal_m4-en_US-20201103-0d0436b.uf2 (nightly build)

tannewt, 2020-11-03:
Thanks @diomark. I'm looking at it now.

tannewt, 2020-11-03:
@diomark I am seeing the `Failed to send 3 bytes` message but then it attempts again and works. (I'm using the weather example from here: https://learn.adafruit.com/weather-display-matrix/code-the-weather-display-matrix ) Are you seeing any successful fetches?

diomark, 2020-11-03:
> @diomark I am seeing the `Failed to send 3 bytes` message but then it attempts again and works. (I'm using the weather example from here: https://learn.adafruit.com/weather-display-matrix/code-the-weather-display-matrix ) Are you seeing any successful fetches?

I tried it twice and received the reload error both times after the intiial weather bpm was displayed. I'll reload and try again. (currently have the quote from io.adafruit.com code loaded)

diomark, 2020-11-03:
loaded this time.. but.. it's just flakey..  log attached. (btw router is like 5ft from this device..)

```

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
255
Jumper set to metric
Getting weather for Los Angeles, US
Set icon to None
gfx loaded
Getting time from internet!
Connecting to AP MVn
Could not connect to internet ('No such ssid', b'MVn')
Retrying in 3 seconds...
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
Time request:  https://io.adafruit.com/api/v2/diomark/integrations/time/strftime?x-aio-key=aio_cOpE64KQPeIhHtiIwJB2QjnfPhyd&tz=America/Los_Angeles&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z
Time reply:  2020-11-03 13:27:10.501 308 2 -0800 PST
Free mem:  31088
Retrieving data...Socket missing recv_into. Using more memory to be compatible
Reply is OK!
{'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604412894, 'country': 'US', 'id': 3694, 'sunset': 1604451485}, 'base': 'stations', 'main': {'temp_min': 21, 'pressure': 1015, 'feels_like': 23.94, 'humidity': 53, 'temp_max': 26.67, 'temp': 24.32}, 'visibility': 6437, 'id': 5368361, 'clouds': {'all': 1}, 'coord': {'lon': -118.24, 'lat': 34.05}, 'name': 'Los Angeles', 'cod': 200, 'weather': [{'id': 721, 'icon': '50d', 'main': 'Haze', 'description': 'haze'}], 'dt': 1604438788, 'wind': {'speed': 2.41, 'deg': 237}}
Response is {'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604412894, 'country': 'US', 'id': 3694, 'sunset': 1604451485}, 'base': 'stations', 'main': {'temp_min': 21, 'pressure': 1015, 'feels_like': 23.94, 'humidity': 53, 'temp_max': 26.67, 'temp': 24.32}, 'visibility': 6437, 'id': 5368361, 'clouds': {'all': 1}, 'coord': {'lon': -118.24, 'lat': 34.05}, 'name': 'Los Angeles', 'cod': 200, 'weather': [{'id': 721, 'icon': '50d', 'main': 'Haze', 'description': 'haze'}], 'dt': 1604438788, 'wind': {'speed': 2.41, 'deg': 237}}
Set icon to 50d
Los Angeles, US
24.32
Haze
53
2.41

```

diomark, 2020-11-03:
reset - log from next run (struggles to get on wifi, then lots of using more memory to be compatible messages)

```
Could not connect to internet ('No such ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Could not connect to internet ('Failed to connect to ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Could not connect to internet ('Failed to connect to ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Could not connect to internet ('Failed to connect to ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Could not connect to internet ('Failed to connect to ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Could not connect to internet ('Failed to connect to ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Could not connect to internet ('Failed to connect to ssid', b'MVn')
Retrying in 3 seconds...
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
Time request:  https://io.adafruit.com/api/v2/diomark/integrations/time/strftime?x-aio-key=aio_cOpE64KQPeIhHtiIwJB2QjnfPhyd&tz=America/Los_Angeles&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z
Time reply:  2020-11-03 13:30:40.210 308 2 -0800 PST
Free mem:  39184
Retrieving data...Socket missing recv_into. Using more memory to be compatible
Reply is OK!
{'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604412894, 'country': 'US', 'id': 3694, 'sunset': 1604451485}, 'base': 'stations', 'main': {'temp_min': 21, 'pressure': 1015, 'feels_like': 23.94, 'humidity': 53, 'temp_max': 26.67, 'temp': 24.32}, 'visibility': 6437, 'id': 5368361, 'clouds': {'all': 1}, 'coord': {'lon': -118.24, 'lat': 34.05}, 'name': 'Los Angeles', 'cod': 200, 'weather': [{'id': 721, 'icon': '50d', 'main': 'Haze', 'description': 'haze'}], 'dt': 1604438788, 'wind': {'speed': 2.41, 'deg': 237}}
Response is {'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604412894, 'country': 'US', 'id': 3694, 'sunset': 1604451485}, 'base': 'stations', 'main': {'temp_min': 21, 'pressure': 1015, 'feels_like': 23.94, 'humidity': 53, 'temp_max': 26.67, 'temp': 24.32}, 'visibility': 6437, 'id': 5368361, 'clouds': {'all': 1}, 'coord': {'lon': -118.24, 'lat': 34.05}, 'name': 'Los Angeles', 'cod': 200, 'weather': [{'id': 721, 'icon': '50d', 'main': 'Haze', 'description': 'haze'}], 'dt': 1604438788, 'wind': {'speed': 2.41, 'deg': 237}}
Set icon to 50d
Los Angeles, US
24.32
Haze
53
2.41
```

diomark, 2020-11-03:
also - here's the results from the adafruit_matrixportal (this one throws errors directly in adafruit_requests.py)

```
Could not connect to internet ('No such ssid', b'MVn')
Retrying in 3 seconds...
Preloading font glyphs: 0123456789
obtaining time from adafruit.io server...
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
New Hour, fetching new data...
Retrieving data...Traceback (most recent call last):
  File "code.py", line 99, in <module>
  File "adafruit_matrixportal/network.py", line 393, in fetch
  File "adafruit_requests.py", line 636, in get
  File "adafruit_requests.py", line 540, in request
  File "adafruit_requests.py", line 537, in request
  File "adafruit_requests.py", line 458, in _send_request
  File "adafruit_requests.py", line 449, in _send
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 82, in send
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 710, in socket_write
RuntimeError: Failed to send 3 bytes (sent 0)
```

tannewt, 2020-11-03:
The `Socket missing recv_into. Using more memory to be compatible` error can be ignored. I added hoping someone would see it and add recv_into. :-)

Are you far from your access point? I think my example may be more reliable because I'm near mine. It usually gets on the WiFi with the first try.

diomark, 2020-11-03:
second run from Matrix_Portal_Learn_Stats

```
Could not connect to internet ('No such ssid', b'MVn')
Retrying in 3 seconds...
Preloading font glyphs: 0123456789
obtaining time from adafruit.io server...
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
New Hour, fetching new data...
Retrieving data...Traceback (most recent call last):
  File "code.py", line 99, in <module>
  File "adafruit_matrixportal/network.py", line 393, in fetch
  File "adafruit_requests.py", line 636, in get
  File "adafruit_requests.py", line 540, in request
  File "adafruit_requests.py", line 537, in request
  File "adafruit_requests.py", line 458, in _send_request
  File "adafruit_requests.py", line 449, in _send
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 82, in send
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 710, in socket_write
RuntimeError: Failed to send 3 bytes (sent 0)

Code done running. Waiting for reload.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to relosoft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Set background to  0
Connecting to AP MVn
Could not connect to internet ('No such ssid', b'MVn')
Retrying in 3 seconds...
Connecting to AP MVn
Preloading font glyphs: 0123456789
obtaining time from adafruit.io server...
Getting time for timezone America/Los_Angeles
Unable to obtain time from Adafruit IO, retrying -  ESP32 not responding
obtaining time from adafruit.io server...
Connecting to AP MVn
Could not connect to internet ('No such ssid', b'MVn')
Retrying in 3 seconds...
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
New Hour, fetching new data...
Retrieving data...Traceback (most recent call last):
  File "code.py", line 99, in <module>
  File "adafruit_matrixportal/network.py", line 393, in fetch
  File "adafruit_requests.py", line 636, in get
  File "adafruit_requests.py", line 540, in request
  File "adafruit_requests.py", line 537, in request
  File "adafruit_requests.py", line 458, in _send_request
  File "adafruit_requests.py", line 449, in _send
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 82, in send
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 710, in socket_write
RuntimeError: Failed to send 3 bytes (sent 0)

Press any key to enter the REPL. Use CTRL-D to reload.
```

diomark, 2020-11-03:
> The `Socket missing recv_into. Using more memory to be compatible` error can be ignored. I added hoping someone would see it and add recv_into. :-)
> 
> Are you far from your access point? I think my example may be more reliable because I'm near mine. It usually gets on the WiFi with the first try.

Nope. literally working on the same desk as the wifi router.. (they're 4-5 feet away from each other)

tannewt, 2020-11-03:
What brand of router is it? I'm not that close. :-) Mine is in a room below mine.

I've seen something similar to your output where it warns that it couldn't connect but then continues to try and make a request. That's weird to me.

diomark, 2020-11-03:
Eero pro. hardwired to ethernet. (no issues w/ other iot devices (mostly esp8266's) around here. they usually connect first try..)

diomark, 2020-11-03:
Putting the can't find ssid issue aside, the Matrix_Portal_Learn_Stats is consistently failing with the same error once it does connect.

```
Preloading font glyphs: 0123456789
obtaining time from adafruit.io server...
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
New Hour, fetching new data...
Retrieving data...Traceback (most recent call last):
  File "code.py", line 99, in <module>
  File "adafruit_matrixportal/network.py", line 393, in fetch
  File "adafruit_requests.py", line 636, in get
  File "adafruit_requests.py", line 540, in request
  File "adafruit_requests.py", line 537, in request
  File "adafruit_requests.py", line 458, in _send_request
  File "adafruit_requests.py", line 449, in _send
  File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 82, in send
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 710, in socket_write
RuntimeError: Failed to send 3 bytes (sent 0)

Press any key to enter the REPL. Use CTRL-D to reload.
```

diomark, 2020-11-03:
Back in the weather app - clean-runs (two times in a row) - but - slowed down significantly because of the socket missing recv_into using more memory to be compatible issue.. Is there a way to tell it to use more memory to start with?

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
255
Jumper set to metric
Getting weather for San Francisco, US
Set icon to None
gfx loaded
Getting time from internet!
Connecting to AP MVn
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
Time request:  https://io.adafruit.com/api/v2/diomark/integrations/time/strftime?x-aio-key=<REMOVED>&tz=America/Los_Angeles&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z
Time reply:  2020-11-03 14:05:29.383 308 2 -0800 PST
Free mem:  31072
Retrieving data...Socket missing recv_into. Using more memory to be compatible
Reply is OK!
{'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604414265, 'country': 'US', 'id': 5817, 'sunset': 1604452121}, 'base': 'stations', 'main': {'temp_min': 13.89, 'pressure': 1018, 'feels_like': 12.4, 'humidity': 55, 'temp_max': 20, 'temp': 16.56}, 'visibility': 10000, 'id': 5391959, 'clouds': {'all': 40}, 'coord': {'lon': -122.42, 'lat': 37.77}, 'name': 'San Francisco', 'cod': 200, 'weather': [{'id': 802, 'icon': '03d', 'main': 'Clouds', 'description': 'scattered clouds'}], 'dt': 1604440848, 'wind': {'speed': 5.1, 'deg': 320}}
Response is {'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604414265, 'country': 'US', 'id': 5817, 'sunset': 1604452121}, 'base': 'stations', 'main': {'temp_min': 13.89, 'pressure': 1018, 'feels_like': 12.4, 'humidity': 55, 'temp_max': 20, 'temp': 16.56}, 'visibility': 10000, 'id': 5391959, 'clouds': {'all': 40}, 'coord': {'lon': -122.42, 'lat': 37.77}, 'name': 'San Francisco', 'cod': 200, 'weather': [{'id': 802, 'icon': '03d', 'main': 'Clouds', 'description': 'scattered clouds'}], 'dt': 1604440848, 'wind': {'speed': 5.1, 'deg': 320}}
Set icon to 03d
San Francisco, US
16.56
Scattered clouds
55
5.1
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
255
Jumper set to imperial
Getting weather for San Francisco, US
Set icon to None
gfx loaded
Getting time from internet!
Connecting to AP MVn
Getting time for timezone America/Los_Angeles
Socket missing recv_into. Using more memory to be compatible
Time request:  https://io.adafruit.com/api/v2/diomark/integrations/time/strftime?x-aio-key=(removed)&tz=America/Los_Angeles&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z
Time reply:  2020-11-03 14:06:42.284 308 2 -0800 PST
Free mem:  31072
Retrieving data...Socket missing recv_into. Using more memory to be compatible
Reply is OK!
{'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604414265, 'country': 'US', 'id': 5817, 'sunset': 1604452121}, 'base': 'stations', 'main': {'temp_min': 57, 'pressure': 1018, 'feels_like': 54.28, 'humidity': 55, 'temp_max': 68, 'temp': 61.77}, 'visibility': 10000, 'id': 5391959, 'clouds': {'all': 40}, 'coord': {'lon': -122.42, 'lat': 37.77}, 'name': 'San Francisco', 'cod': 200, 'weather': [{'id': 802, 'icon': '03d', 'main': 'Clouds', 'description': 'scattered clouds'}], 'dt': 1604441005, 'wind': {'speed': 11.41, 'deg': 320}}
Response is {'timezone': -28800, 'sys': {'type': 1, 'sunrise': 1604414265, 'country': 'US', 'id': 5817, 'sunset': 1604452121}, 'base': 'stations', 'main': {'temp_min': 57, 'pressure': 1018, 'feels_like': 54.28, 'humidity': 55, 'temp_max': 68, 'temp': 61.77}, 'visibility': 10000, 'id': 5391959, 'clouds': {'all': 40}, 'coord': {'lon': -122.42, 'lat': 37.77}, 'name': 'San Francisco', 'cod': 200, 'weather': [{'id': 802, 'icon': '03d', 'main': 'Clouds', 'description': 'scattered clouds'}], 'dt': 1604441005, 'wind': {'speed': 11.41, 'deg': 320}}
Set icon to 03d
San Francisco, US
61.77
Scattered clouds
55
11.41
```

tannewt, 2020-11-03:
It will use more memory when it prints that message.

I'm trying to add debug prints to the nina firware on the ESP32 but am having trouble getting it going.

diomark, 2020-11-03:
tnx Scott - let me know f you'd like me to test anything.

diomark, 2020-11-03:
Weather sketch has been running for a while; now I'm seeing the failed to send 3 bytes message..
[image]

tannewt, 2020-11-04:
@diomark Does it always recover? This can happen when the socket is closed before we send.

tannewt, 2020-11-05:
@diomark Please try #114 in conjunction with adafruit_requests.py from here: https://github.com/tannewt/Adafruit_CircuitPython_Requests/tree/better_error_handling

diomark, 2020-11-11:
The new libraries at adafruit-circuitpython-bundle-6.x-mpy-20201111 have been much more stable.  Matrix_Portal_Learn_Stats is now working.

tannewt, 2020-11-12:
Ok, great! Closing the issue. If folks find further problems, please open a new issue.

closed 2020-11-12 by tannewt