adafruit/Adafruit_CircuitPython_ESP32SPI #116: MatrixPortal won't soft reboot using esp.reset()

astrilchuk, 2020-11-16:
Using firmware version 1.7.1 and CircuitPython 6.0.0 rc2 I'm not able to soft reboot via esp.reset().  Example code is shown below.  It's as if esp.reset() simply acts as a no-op.

I am occasionally getting an 'ESP not responding' error which I'm trying to consistently reproduce.  But in the meantime I'm not able to recover from the error by forcing a reset.

```
from adafruit_esp32spi import adafruit_esp32spi
import digitalio
import busio
import board

esp32_ready = digitalio.DigitalInOut(board.ESP_BUSY)
esp32_gpio0 = digitalio.DigitalInOut(board.ESP_GPIO0)
esp32_reset = digitalio.DigitalInOut(board.ESP_RESET)
esp32_cs = digitalio.DigitalInOut(board.ESP_CS)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)

esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset, esp32_gpio0)

firmware_version = str(esp.firmware_version, "utf-8")

print(f"ESP Firmware Version: {firmware_version}")

print("resetting...")
esp.reset()
print("done resetting.")
```

tannewt, 2020-11-16:
What do you mean by soft reboot? esp.reset() won't control anything about CircuitPython.

astrilchuk, 2020-11-16:
Ya, I'm starting to understand that is only the co-processor.  I read in the CircuitPython docs that esp.reset() "Hard reset the ESP32 using the reset pin".  So I assumed that meant the whole board.  How would I perform a board-level reset?

I've run into the 'ESP not responding' error dozens of times and I'm at wit's end trying just simply recover from it (having said that, my latest tests seem promising).

Micropython had machine.reset() and machine.soft_reset() but the machine namespace doesn't exist in CircuitPython.  Is there a CircuitPython equivalent?

tannewt, 2020-11-17:
You can do [`microcontroller.reset()`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/microcontroller/index.html#microcontroller.reset) but this isn't board level. That will only reset the SAMD51. The soft_reset equivalent is `supervisor.reload()`.

I'd suggest increasing the ESP32SPI not responding timeout here: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/adafruit_esp32spi/adafruit_esp32spi.py#L183

Are you getting the error while making a request? If so, what url are you connected to?

astrilchuk, 2020-11-18:
Thanks for that, I'll have a look at those methods.

Fortunately, with my latest tests, I've been able to recover from an MQTT/"ESP Not Responding" error by performing an esp.reset(), reconnecting to wifi, and finally reconnecting back to the MQTT server.  This has alleviated my need for a hard or soft reset.

I am connected to a local MQTT instance and I believe the error will occur when the MQTT ping is sent (probably occurs with other data sent but the ping is the most frequent).  The error will be either "failed to send 2 bytes" or "ESP not responding".  Before I knew how to recover from the error, only a manual reset would recover.

closed 2020-11-18 by astrilchuk