adafruit/Adafruit_CircuitPython_ESP32SPI #120: Documentation on status LED?

ajs256, 2020-12-26:
I can't seem to find documentation on how this library controls the NeoPixel on my MatrixPortal. Could someone please point me to where I could find that?

jerryneedell, 2020-12-26:
This is probably not what you were asking for but hopefully it will help.
It is used by the Wifi manager  https://learn.adafruit.com/adafruit-pyportal/internet-connect#wifi-manager-3034791-36 , but you are correct, the actual usage is not documented. The simplest thing now would be to look at the code for places where the pixel_status is set -- for example 
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/adafruit_esp32spi/adafruit_esp32spi_wifimanager.py#L139
and
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/master/adafruit_esp32spi/adafruit_esp32spi_wifimanager.py#L142

show it being set red  (100,0,0) before then green(0,100,0) after "connecting".

FoamyGuy, 2021-12-30:
closed by #148

closed 2021-12-30 by FoamyGuy