adafruit/Adafruit_CircuitPython_ESP32SPI #121: __init__ arguments are not Pins

dhalbert, 2020-12-28:
The arg names here are suffixed as `_pin`, but they should actually be `DigitalInOut` objects. The arg names could be clearer, maybe `_dio` or just drop the suffix. At least one user might have been confused by this: https://forums.adafruit.com/viewtopic.php?f=60&t=173353.

https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/fce466bd2bb70ca86b79e5cb36bbaca00afacfd1/adafruit_esp32spi/adafruit_esp32spi.py#L136-L138

brentru, 2020-12-28:
@dhalbert  I think dropping the suffix would be clearer compared to substituting `_pin` for  `_dio`

tekktrik, 2021-11-01:
Is this something we want to do?  I'm happy to submit a PR for this :) Was also planning to add type annotations, if that helps at all.

FoamyGuy, 2022-01-24:
resolved by #146

closed 2022-01-24 by FoamyGuy

FoamyGuy, 2022-01-24:
@tekktrik all of the other instances of similar `*_pin` named arguments are linked above here in this issue now.

tannewt, 2022-01-24:
I agree with @brentru that dropping the suffix is clearer than changing it to `_dio`. Adding the suffix is redundant with the type info.

tekktrik, 2022-01-24:
I can submit another PR to drop it if that's the case

tannewt, 2022-01-25:
I think it's important to point out that by taking DigitalInOut its usually meant to work with GPIO expander provided objects as well. Pin should be only used when a native pin object is expected.