adafruit/Adafruit_CircuitPython_ESP32SPI #123: Can not get it to work on Feather M4 + AirLift: RuntimeError: Error response to command

otherguy, 2021-01-07:
I have a Feather M4 Express with an AirLift FeatherWing. They are not stacked but sit next to each other on a breadboard. They share a common ground and get +5V through the VUSB pin. I believe I have connected all required (and more) pins: SCK, MOSI, MISO, RX, TX, as well as D10, D11, D12, D13.

I'm running CircuitPython 6.0.1 (or 6.1.0rc0, same error there) and the latest libraries and bootloader v3.10.0.

This simple program fails:

```python
from adafruit_esp32spi import adafruit_esp32spi
from digitalio import DigitalInOut
from busio import SPI
import board 

esp32_cs    = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

spi = SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

if esp.status == adafruit_esp32spi.WL_IDLE_STATUS:
  print("ESP32 Firmware v{}, MAC: {}".format(esp.firmware_version.decode('ascii'), ':'.join('%02X' % b for b in esp.MAC_address_actual)))

  for ap in esp.scan_networks():
    print("\t%s\t\tRSSI: %d" % (str(ap['ssid'], 'utf-8'), ap['rssi']))
  else:
    print("ESP not in idle state!")
```

And this is the output:
```
ESP32 Firmware v1.2.2, MAC: B8:F0:09:95:BD:7C
Traceback (most recent call last):
  File "code.py", line 466, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 402, in scan_networks
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 379, in get_scan_networks
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 308, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 292, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 268, in _wait_spi_char
RuntimeError: Error response to command
```

So clearly the Feather M4 can communicate with the AirLift featherwing (it retrieves the firmware version and MAC) but I can't execute any further commands.

brentru, 2021-03-05:
Could you try upgrading your ESP32 firmware to the latest version of the nina firmware?
https://learn.adafruit.com/upgrading-esp32-firmware/upgrade-an-external-esp32#upload-serial-passthrough-code-for-feather-or-itsybitsy-3055058-4

brentru, 2021-03-26:
@otherguy  Have you tried the steps above?

otherguy, 2021-05-05:
@brentru sorry that took me a while, the M4 was in use and I didn't want to take the project down until recently.

Seems I wired everything correctly because all I did was close the solder bridges on the ESP32 / Airlift and upload the latest firmware:

```
esptool.py --port /dev/tty.usbmodem14201 --before no_reset --baud 115200 write_flash 0 ~/Downloads/NINA_W102-1.7.3.bin
esptool.py v3.0
Serial port /dev/tty.usbmodem14201
Connecting........_____....._____...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b8:f0:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 1159168 bytes to 633551...
Wrote 1159168 bytes (633551 compressed) at 0x00000000 in 211.7 seconds (effective 43.8 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
```

After that, I ran the example code:

```
code.py output:
ESP32 SPI webclient test
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.7.3\x00')
MAC addr: ['0x7c', '0xbd', '0x95', '0x9', '0xf0', '0xb8']
        99 problems but WiFi ain't one          RSSI: -48
        MicroPython-3b0408              RSSI: -65
        UPC Wi-Free             RSSI: -73
Connecting to AP...
Connected to 99 problems but WiFi ain't one     RSSI: -46
My IP address is 172.16.1.132
IP lookup adafruit.com: 104.20.38.240
Ping google.com: 40 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)
----------------------------------------
Done!
```

Looks like it went from 0 to 100 with that new firmware! Thanks for all your help!

It's strange that a slightly older firmware could cause something like this though.

I'm closing this but feel free to reply!

closed 2021-05-05 by otherguy