adafruit/Adafruit_CircuitPython_ESP32SPI #134: Only allows one TLS connection at a time

askpatrickw, 2021-04-15:
In [release 3.5.3](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/releases/tag/3.5.3) a change was made: 
"Only allow one TLS connection at a time. Nina FW doesn't work well with multiple."

This broke the Azure_IOT workflow where you use HTTP to determiner your Iot Hub and then connect to it over MQTT.
https://github.com/adafruit/Adafruit_CircuitPython_AzureIoT/pull/31#issuecomment-819900695

There was a workaround for that library, but I wanted to document the issue.

justmobilize, 2024-05-23:
@tannewt do you remember (almost 4 years ago), why this was removed?

Testing, using:
```py
from digitalio import DigitalInOut
import board
import os

import adafruit_connection_manager
from adafruit_esp32spi import adafruit_esp32spi

esp32_cs = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

spi = board.SPI()
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

# Get WiFi details, ensure these are setup in settings.toml
ssid = os.getenv("WIFI_SSID")
password = os.getenv("WIFI_PASSWORD")

print(f"\nConnecting to {ssid}...")
esp.connect(ssid, password)
print("connected")

pool = adafruit_connection_manager.get_radio_socketpool(esp)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(esp)

connection_manager = adafruit_connection_manager.get_connection_manager(pool)
socket_1 = connection_manager.get_socket("io.adafruit.com", 443, "https:", is_ssl=True, ssl_context=ssl_context)
socket_2 = connection_manager.get_socket("www.adafruit.com", 443, "https:", is_ssl=True, ssl_context=ssl_context)
```

I get:
```console
>>> socket_1 = connection_manager.get_socket("io.adafruit.com", 443, "https:", is_ssl=True, ssl_context=ssl_context)
>>> socket_2 = connection_manager.get_socket("www.adafruit.com", 443, "https:", is_ssl=True, ssl_context=ssl_context)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_connection_manager.py", line 332, in get_socket
  File "adafruit_connection_manager.py", line 244, in _get_connected_socket
  File "adafruit_connection_manager.py", line 61, in connect
  File "adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 114, in connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 836, in socket_connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 721, in socket_open
OSError: 23
```

If I comment out the code to only allow 1 I get:
```console
>>> socket_1 = connection_manager.get_socket("io.adafruit.com", 443, "https:", is_ssl=True, ssl_context=ssl_context)
>>> socket_2 = connection_manager.get_socket("www.adafruit.com", 443, "https:", is_ssl=True, ssl_context=ssl_context)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_connection_manager.py", line 332, in get_socket
  File "adafruit_connection_manager.py", line 244, in _get_connected_socket
  File "adafruit_connection_manager.py", line 61, in connect
  File "/lib/adafruit_esp32spi/adafruit_esp32spi_socketpool.py", line 114, in connect
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 836, in socket_connect
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 732, in socket_open
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 341, in _send_command_get_response
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 311, in _wait_response_cmd
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 293, in _check_data
BrokenPipeError: Expected 01 but got 00
```

cc @dhalbert

dhalbert, 2024-05-24:
I found a discussion in the forums that is at the same time as the 3.5.3 release: https://forums.adafruit.com/viewtopic.php?p=836698

dhalbert, 2024-05-24:
Given this testing, I'm not sure this can be fixed now. We can revisit after https://github.com/adafruit/nina-fw/pull/62 is done. In the meantime, I'll merge #207.