adafruit/Adafruit_CircuitPython_ESP32SPI #137: RGB LED doesn't turn off on arduino nano connect 2040

Neradoc, 2021-08-14:
The following code on the Arduino Nano Connect 2040 results in a dim white light:
```py
import time
import board
import busio
from digitalio import DigitalInOut
from adafruit_esp32spi import adafruit_esp32spi
esp32_cs = DigitalInOut(board.CS1)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)
spi = busio.SPI(board.SCK1, board.MOSI1, board.MISO1)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

# The LED-related part:

import adafruit_rgbled
from adafruit_esp32spi import PWMOut

LEDR = 27
LEDG = 25
LEDB = 26

RED_LED = PWMOut.PWMOut(esp, LEDR)
GREEN_LED = PWMOut.PWMOut(esp, LEDG)
BLUE_LED = PWMOut.PWMOut(esp, LEDB)
status_light = adafruit_rgbled.RGBLED(RED_LED, GREEN_LED, BLUE_LED, invert_pwm=True)

status_light.color = (0,0,0)
```
To be able to turn it off, I used `esp.set_digital_write` and ended up changing the PWM pin to digital when the duty cycle is 0 or 100% like this:
```py
class PWMOut2(PWMOut.PWMOut):
    @property
    def duty_cycle(self):
        return super().duty_cycle()

    @duty_cycle.setter
    def duty_cycle(self, duty_cycle):
        self._is_deinited()
        if not isinstance(duty_cycle, (int, float)):
            raise TypeError("Invalid duty_cycle, should be int or float.")
        duty_cycle /= 65535.0
        if not 0.0 <= duty_cycle <= 1.0:
            raise ValueError("Invalid duty_cycle, should be between 0.0 and 1.0")
        # If the value is 0 or 1, turn into a digital pin instead of PWM
        if duty_cycle == 0.0:
            self._esp.set_pin_mode(self._pwm_pin,1)
            self._esp.set_digital_write(self._pwm_pin, False)
        elif duty_cycle == 1.0:
            self._esp.set_pin_mode(self._pwm_pin,1)
            self._esp.set_digital_write(self._pwm_pin, True)
        else:
            self._esp.set_analog_write(self._pwm_pin, duty_cycle)

RED_LED = PWMOut2(esp, LEDR)
GREEN_LED = PWMOut2(esp, LEDG)
BLUE_LED = PWMOut2(esp, LEDB)
status_light = adafruit_rgbled.RGBLED(RED_LED, GREEN_LED, BLUE_LED, invert_pwm=True)

status_light.color = (0,0,0)
```
This might be an issue with the Nina firmware or the board's design, but I wonder if it would be ok to have that kind of things in the base code, or if it would be better to make a community library for the Nano.

anecdata, 2021-08-14:
There was a PWM fix in the Adafruit fork of NINA <https://github.com/adafruit/nina-fw/pull/33>, that is not in the Arduino version.

anecdata, 2022-01-07:
Is there a way to re-file an issue to the parent repository, or push a PR up to the parent?

FoamyGuy, 2022-01-07:
@anecdata there is a way to transfer an issue to a different repo. I'm not sure what you mean to re-file it in the parent repository, if transfer would work we can do that though

anecdata, 2022-01-07:
It seems that the issue is really with the Arduino NINA firmware (parent of the forked Adafruit NINA firmware). Ideally we could push the Adafruit NINA PR up to Arduino?