adafruit/Adafruit_CircuitPython_ESP32SPI #138: Missing Type Annotations

FoamyGuy, 2021-09-27:
There are missing type annotations for some functions in this library.

The `typing` module does not exist on CircuitPython devices so the import needs to be wrapped in try/except to catch the error for missing import. There is an example of how that is done here:

```python
try:
    from typing import List, Tuple
except ImportError:
    pass
```

Once imported the typing annotations for the argument type(s), and return type(s) can be added to the function signature. Here is an example of a function that has had this done already:

```python
def wrap_text_to_pixels(
    string: str, max_width: int, font=None, indent0: str = "", indent1: str = ""
) -> List[str]:
```
If you are new to Git or Github we have a guide about contributing to our projects here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

There is also a guide that covers our CI utilities and how to run them locally to ensure they will pass in Github Actions here: https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/check-your-code In particular the pages: `Sharing docs on ReadTheDocs` and `Check your code with pre-commit` contain the tools to install and commands to run locally to run the checks.

If you are attempting to resolve this issue and need help, you can post a comment on this issue and tag both @foamyguy and @kattni or reach out to us on Discord: https://adafru.it/discord in the `#circuitpython-dev` channel.

The following locations are reported by mypy to be missing type annotations:

- [ ] adafruit_esp32spi/digitalio.py:43
- [ ] adafruit_esp32spi/digitalio.py:50
- [ ] adafruit_esp32spi/digitalio.py:64
- [ ] adafruit_esp32spi/digitalio.py:118
- [ ] adafruit_esp32spi/digitalio.py:126
- [ ] adafruit_esp32spi/digitalio.py:133
- [ ] adafruit_esp32spi/digitalio.py:142
- [ ] adafruit_esp32spi/digitalio.py:156
- [ ] adafruit_esp32spi/digitalio.py:176
- [ ] adafruit_esp32spi/digitalio.py:195
- [ ] adafruit_esp32spi/PWMOut.py:29
- [ ] adafruit_esp32spi/PWMOut.py:44
- [ ] adafruit_esp32spi/PWMOut.py:69
- [ ] adafruit_esp32spi/PWMOut.py:89
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:136
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:195
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:247
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:254
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:262
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:273
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:279
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:314
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:407
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:413
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:419
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:425
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:431
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:443
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:449
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:514
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:519
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:551
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:592
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:596
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:601
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:614
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:637
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:670
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:681
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:685
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:728
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:737
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:753
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:775
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:787
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:802
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:808
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:815
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:832
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:845
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:859
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:878
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:913
- [ ] adafruit_esp32spi/adafruit_esp32spi.py:935
- [ ] adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py:39
- [ ] adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py:49
- [ ] adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py:70
- [ ] adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py:108
- [ ] adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py:166
- [ ] adafruit_esp32spi/adafruit_esp32spi_wsgiserver.py:182
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:33
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:210
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:229
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:247
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:266
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:285
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:304
- [ ] adafruit_esp32spi/adafruit_esp32spi_wifimanager.py:330
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:24
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:38
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:55
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:67
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:80
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:89
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:111
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:162
- [ ] adafruit_esp32spi/adafruit_esp32spi_socket.py:169

tekktrik, 2022-03-25:
I really want to tackle this, but I just want to note here before I forget that we may want to consider adding some `typing_extensions.Protocol` types to `circuitpython_typing` to help with some of the typing in this library!

dhalbert, 2023-01-06:
The lack of annotations and param documentation causes it to be non-obvious that ` ESP_SPIcontrol()` takes `DigitalInOut`s, not `Pin`s. This caught at least one user: https://github.com/adafruit/circuitpython/issues/7386#issuecomment-1373110653

tekktrik, 2023-01-06:
I can take this one on then, it seems I said I would a while ago :sweat_smile:

pedrovs16, 2023-04-24:
Working on it during PyCon 2023 sprints.