adafruit/Adafruit_CircuitPython_ESP32SPI #143: Getting "ESP32 timed out on SPI select", then "sRepeated socket failure" for retries

bembengarifin, 2021-11-18:
Hi,

I have a matrix portal that polls some REST API every few minutes.

It works fine but after leaving it running for few hours for few times, I noticed that it always stops working with this error "ESP32 timed out on SPI select" then further attempts (code will retry in another minute) will return "Repeated socket failure"

I use these artefacts below:
adafruit-circuitpython-matrixportal_m4-en_GB-7.0.0.uf2
adafruit-circuitpython-bundle-7.x-mpy-20211101.zip

I saw similar error reported but I assume it's already fixed in circuit python 7 -> https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/issues/102

I verified that my firmware version is 1.2.2
ESP32 SPI webclient test
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.2.2\x00')

It seems that the latest version is 1.3.0? 
Tried to download the MatrixPortal_ESP32_Passthru.UF2 (from https://learn.adafruit.com/upgrading-esp32-firmware/upgrade-an-airlift-all-in-one-board) but then it doesn't seem to do anything so rolled it back to adafruit-circuitpython-matrixportal_m4-en_GB-7.0.0.uf2

Let me know if any further details needed to troubleshoot this issue.

Thanks

anecdata, 2021-11-18:
Latest NINA firmware is 1.7.4, there have been some improvements since 1.2.2. You could try to get help on Discord for updating NINA, and for discussion of making code more resilient.

Networking errors will happen. I see the same exceptions that you are reporting, but retrying often fixes it. Some things you can try in your code., kind of in escalating order:

- try to verify that the network is solid... e.g., do other devices in similar proximity to the Access point ever have issues? is the RSSI good from the Station perspective? From the AP perspective? Is the LAN and internet access generally robust?
- retry the Request several times
- you could try increasing the timeout a little, or increasing waits between retries
- `esp.reset` will put the ESP32 co-processor in a good fresh state, it cures many ills if the above don't do it

If all else fails, you can `supervisor.reload()`, or worst case: `mircocontroller.reset()`.

bembengarifin, 2021-11-19:
thanks for the quick and detailed response.

I managed to get 1.7.4 installed after "carefully" going through the instruction again -> https://learn.adafruit.com/upgrading-esp32-firmware/upgrade-an-airlift-all-in-one-board

The internet access/network should be fine, the matrix display is just next to the router.

I will give it a try again and follow your other recommendations in case it still fails, thanks again

closed 2021-11-19 by bembengarifin