adafruit/Adafruit_CircuitPython_ESP32SPI #144: Feather M4 Express not connecting with Air Lift ESP32 Coprocessor after update to NINA_W102-1.7.4

LAWBear, 2021-11-24:
Purchased Feather M4 Express to allow use of Air Lift ESP32 Coprocessor breakout board via SPI. 
Previous test with original Air Lift firmware using Feather M0 Adalogger were successful using WIFINA examples.  

Following detailed instructions, I updated firmware on Air Lift using SerialPassThru on new Feather M4 Express, and reflashed Air Lift with NINAW102-1.7.4 as clear instructions provided:
[image]
The above output assumed success of the update as instructed, however EXTENSIVE testing of various CircuitPython examples from Python 7 bundles all fail with similar errors (timed out waiting for SPI char) as follows:
ESP32 SPI webclient test
Traceback (most recent call last):
  File "code.py", line 50, in <module>
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 292, in _wait_response_cmd
  File "/lib/adafruit_esp32spi/adafruit_esp32spi.py", line 271, in _wait_spi_char
RuntimeError: Timed out waiting for SPI char

I am desperate for a kind wizard to advise my errors and lead to my hope of implementing the BLE capabilities of the Feather M4 Express with Air Lift ESP32 Coprocessor breakout board.

anecdata, 2021-11-24:
Should work, I have an identical configuration, but I think this is more of a support issue. Can you post about it on the Adafruit Discord #help-with-CircuitPython, (or if not that then in the Adafruit support forums)?

JayToltTech, 2021-12-05:
@LAWBear I have an identical configuration and I had your problem *until* I flashed to W102-1.7.4 which fixed the problem.

My suspicion would be that your flash didn't work as expected.  I had to flash a few times to get it right, though I don't think I saw a 'false positive' where it said it flashed but didn't like seems to be in your case.

anecdata, 2021-12-05:
And just to be squeaky-clean about it, you can use the `erase_flash` option before the flashing step.

LAWBear, 2021-12-06:
Many thanks to the wizards, I will implement all checks and suggestions soonest.

dhalbert, 2022-06-06:
Closing as possibly resolved, please reopen if not.

closed 2022-07-20 by dhalbert