adafruit/Adafruit_CircuitPython_ESP32SPI #154: TypeError when trying to use requests.get

jasonwashburn, 2022-01-22:
Any time I try to use the requests library on my Pyportal since upgrading to CircuitPython 7.1.1 I get the following error:
```
code.py output:
Connecting to AP...
Connected to WBurn-2G 	RSSI: -50
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
Traceback (most recent call last):
  File "code.py", line 58, in <module>
  File "adafruit_requests.py", line 847, in get
  File "adafruit_requests.py", line 719, in request
  File "adafruit_requests.py", line 210, in __init__
  File "adafruit_requests.py", line 297, in _readto
  File "adafruit_requests.py", line 241, in _recv_into
TypeError: function takes 2 positional arguments but 3 were given

Code done running.
```

I've tried multiple example files that use the library with the same result. The currently loaded version is  the code below.
```
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

# adafruit_requests usage with an esp32spi_socket
import board
import busio
from digitalio import DigitalInOut
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
from adafruit_esp32spi import adafruit_esp32spi
import adafruit_requests as requests

# Add a secrets.py to your filesystem that has a dictionary called secrets with "ssid" and
# "password" keys with your WiFi credentials. DO NOT share that file or commit it into Git or other
# source control.
# pylint: disable=no-name-in-module,wrong-import-order
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

# If you are using a board with pre-defined ESP32 Pins:
esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

# If you have an externally connected ESP32:
# esp32_cs = DigitalInOut(board.D9)
# esp32_ready = DigitalInOut(board.D10)
# esp32_reset = DigitalInOut(board.D5)

# If you have an AirLift Featherwing or ItsyBitsy Airlift:
# esp32_cs = DigitalInOut(board.D13)
# esp32_ready = DigitalInOut(board.D11)
# esp32_reset = DigitalInOut(board.D12)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

print("Connecting to AP...")
while not esp.is_connected:
    try:
        esp.connect_AP(secrets["ssid"], secrets["password"])
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(esp.ssid, "utf-8"), "\tRSSI:", esp.rssi)

# Initialize a requests object with a socket and esp32spi interface
socket.set_interface(esp)
requests.set_socket(socket, esp)

TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_GET_URL = "https://httpbin.org/get"
JSON_POST_URL = "https://httpbin.org/post"

print("Fetching text from %s" % TEXT_URL)
response = requests.get(TEXT_URL)
print("-" * 40)

print("Text Response: ", response.text)
print("-" * 40)
response.close()

print("Fetching JSON data from %s" % JSON_GET_URL)
response = requests.get(JSON_GET_URL)
print("-" * 40)

print("JSON Response: ", response.json())
print("-" * 40)
response.close()

data = "31F"
print("POSTing data to {0}: {1}".format(JSON_POST_URL, data))
response = requests.post(JSON_POST_URL, data=data)
print("-" * 40)

json_resp = response.json()
# Parse out the 'data' key from json_resp dict.
print("Data received from server:", json_resp["data"])
print("-" * 40)
response.close()

json_data = {"Date": "July 25, 2019"}
print("POSTing data to {0}: {1}".format(JSON_POST_URL, json_data))
response = requests.post(JSON_POST_URL, json=json_data)
print("-" * 40)

json_resp = response.json()
# Parse out the 'json' key from json_resp dict.
print("JSON Data received from server:", json_resp["json"])
print("-" * 40)
response.close()
```

I'm currently running CircuitPython 7.1.1 Stable with the adafruit_requests library from adafruit-circuitpython-requests-7.x-mpy-1.10.5.zip

FoamyGuy, 2022-01-22:
Transferred to ESP32SPI. This is where it will need to be fixed I believe. 

recv_into() declared here:
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/52208e4752bee510b24ac775763776231b5912cc/adafruit_esp32spi/adafruit_esp32spi_socket.py#L164

only has one parameter. But the requests library is written to pass in two arguments here: 
https://github.com/adafruit/Adafruit_CircuitPython_Requests/blob/main/adafruit_requests.py#L241

I think the recv_into() method here will need to accept a second argument for size.

tekktrik, 2022-01-22:
Oops!  Didn't realize it needed a second parameter when I implemented it not long ago.  Thanks y'all, hot fix PR submitted!

FoamyGuy, 2022-01-24:
Resolved by #155

closed 2022-01-24 by FoamyGuy