adafruit/Adafruit_CircuitPython_ESP32SPI #156: Prints password in clear text if debug is True

FoamyGuy, 2022-01-22:
This library prints the wifi password from `secrets.py` in cleartext if debug is True from the code here:
```
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/52208e4752bee510b24ac775763776231b5912cc/adafruit_esp32spi/adafruit_esp32spi.py#L531
```
I can see how this could be helpful for troubleshooting, especially noticing if the password was incorrect. But I think it should require an extra boolean parameter or something to do this. 

I would like to see all the rest of the debugging messages printed, but don't really want my password printed along with it.

rsbohn, 2022-01-22:
Make debug safe for streaming.

dhalbert, 2022-01-22:
How about if debug is 1 or True it doesn't print the password. If debug is >=2, it does print the password.

FoamyGuy, 2022-01-22:
@dhalbert  that sounds like a good solution to me

tannewt, 2022-01-24:
I'd prefer a separate `show_passwords` kwarg. A kwarg is self documenting. A different debug value isn't.

PR #178 (2023-04-27): Don't print password by default in debug
closes #156
closed 2023-04-27 by tannewt