adafruit/Adafruit_CircuitPython_ESP32SPI #158: Could SPI issues be raise as IOErrors?

dannystaple, 2022-03-06:
When there are SPI failures (which I think are mostly down to underpowering the device or needing more ballast capacitors), the system raises a RuntimeError. Just so these can be handled differently (ie got wrong message back, or failed to get acknowledgment), could they be raised as an IOError?

Example:
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/ef26881497c45636fcb3dbdba2f07d91c0dfdb85/adafruit_esp32spi/adafruit_esp32spi.py#L239

This is so they are distinctly different from the socket type errors, like failed to connect to host (less hardware, more wrong hostname/ip):

https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/ef26881497c45636fcb3dbdba2f07d91c0dfdb85/adafruit_esp32spi/adafruit_esp32spi_socket.py#L77

tannewt, 2022-03-07:
IOError is deprecated in CPython. It is now replaced by OSError. I'd be ok switching to it though.

dhalbert, 2022-07-20:
Fixed by #164, I would say, right, @dannystaple ? :)

closed 2022-07-20 by dhalbert