adafruit/Adafruit_CircuitPython_ESP32SPI #162: ESP failures within 1-2 hours

dsohrabian, 2022-03-30:
I am using a Matrix Portal to make a couple post requests every 30 seconds. It sends a 2kb payload, receives a similarly small response, and updates an LED board. It works successfully for some time.  Originally when using a matrixportal.network Network object to connect to internet,   would use adafruit_requests.post method. After about 1-2 hours I would get the following error. The ESP unit stopped responding. 

`File "adafruit_requests.py", line 869, in post
File "adafruit_requests.py", line 720, in request
File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 141, in recv
File "adafruit_esp32spi/adafruit_esp32spi.py", line 748, in socket_read
File "adafruit_esp32spi/adafruit_esp32spi.py", line 326, in _send_command_get_ response
File "adafruit_esp32spi/adafruit_esp32spi.py", line 282, in _wait_response_cmd
File "adafruit_esp32spi/adafruit_esp32spi.py", line 191, in _wait_for_ready
RuntimeError: ESP32 not responding`

I updated the ESP firmware according to Adafruit guide for Matrix Portal and started getting this error too:

`File "adafruit_requests.py", line 869, in post
File "adafruit_requests.py", line 720, in request
File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 138, in recv
File "adafruit_esp32spi/adafruit_esp32spi_socket.py", line 217, in available
File "adafruit_esp32spi/adafruit_esp32spi.py", line 732, in socket_available
File "adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
File "adafruit_esp32spi/adafruit_esp32spi.py", line 239, in _send_command
RuntimeError: ESP32 timed out on SPI select`

Finally, I switched to a WifiManager object (instead of adafruit_requests methods) and thought I was making progress. I set up a try except that resets the wifi. It resets occasionally. But the board freezes entirely after 30-120 minutes, loses USB connection if plugged to PC, and the status light (neopixel) tied to the wifi object stays on permanently, i.e. the unit hangs on code somewhere.?

I don't know how to debug further. It's been weeks of testing. From what I can tell this seems like something isn't timing out as it should, the hardware has an issue, or the firmware is?

dsohrabian, 2022-04-16:
I have run the board with ESP.debug set to True and have some debug logging showing how the ESP hangs. It has not been very insightful to me, but thought it may be helpful. This example ran for 2hr 30 minutes before freezing.

Here is the ESP writing and sending my POST request.
```
Writing: b'POST'
Writing: b' /'
Writing: b'Arrivals.aspx/getStopTimes'
Writing: b' HTTP/1.1\r\n'
Writing: b'Accept-Encoding'
Writing: b': '
Writing: b'gzip, deflate'
Writing: b'\r\n'
Writing: b'Content-Type'
Writing: b': '
Writing: b'application/json'
Writing: b'\r\n'
Writing: b'Host'
Writing: b': '
Writing: b'nextconnect.riderta.com'
Writing: b'\r\n'
Writing: b'User-Agent'
Writing: b': '
Writing: b'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36'
Writing: b'\r\n'
Writing: b'Referer'
Writing: b': '
Writing: b'http://nextconnect.riderta.com/LiveDepartureTimes'
Writing: b'\r\n'
Writing: b'Accept'
Writing: b': '
Writing: b'application/json'
Writing: b'\r\n'
Writing: b'Content-Type: application/json\r\n'
Writing: b'Content-Length: 89\r\n'
Writing: b'\r\n'
Writing: b'{"directionID": 5, "stopID": 12925, "tpID": 0, "useArrivalTimes": "false", "routeID": 60}'
Connection status: 3
```
Here is a successful read sequence after my POST request is written and sent as above.

```
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 1022 bytes available
Reading 1 bytes from ESP socket with status 4
ESPSocket: 1021 bytes available
Reading 32 bytes from ESP socket with status 4
ESPSocket: 989 bytes available
Reading 31 bytes from ESP socket with status 4
ESPSocket: 958 bytes available
Reading 20 bytes from ESP socket with status 4
ESPSocket: 938 bytes available
Reading 14 bytes from ESP socket with status 4
ESPSocket: 924 bytes available
Reading 32 bytes from ESP socket with status 4
ESPSocket: 892 bytes available
Reading 60 bytes from ESP socket with status 4
ESPSocket: 832 bytes available
Reading 12 bytes from ESP socket with status 4
ESPSocket: 820 bytes available
[... 108 lines cut ...]
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 0 bytes available
ESPSocket: 1022 bytes available
Reading 1 bytes from ESP socket with status 4
ESPSocket: 1021 bytes available
Reading 32 bytes from ESP socket with status 4
ESPSocket: 989 bytes available
Reading 31 bytes from ESP socket with status 4
ESPSocket: 958 bytes available
Reading 20 bytes from ESP socket with status 4
```
As you see, the ESP stops responding (?) for some reason in the middle of unpacking the response from the POST request.
Hoping I can find a workaround or help find the bug. Thank you!

dsohrabian, 2022-04-18:
Is there something that I might be doing wrong code-wise that is causing memory issues?
If so, is there a good way to debug to check it out?

dhalbert, 2022-04-18:
I am looking at related issues also, see for instance: https://github.com/adafruit/circuitpython/issues/6205.

I ran a simple get via the requests library every second for several hours without problem. 

If you have not already done so, try updating the ESP32 firmware to 1.7.4. What version are you currently running? It's reported in the "Internet Test" example given in the guide.

dsohrabian, 2022-04-18:
> If you have not already done so, try updating the ESP32 firmware to 1.7.4. What version are you currently running? It's reported in the "Internet Test" example given in the guide.

Thanks for taking a look.  This definitely seems like the same issue so far.

I updated several weeks ago to 1.74 firmware for ESP32.  I will try running your simpler web fetcher and see what messages I get or if it hangs.

dsohrabian, 2022-04-18:
I adapted your code @dhalbert with tweaks of 

- turning the GET into a POST request
- calls .json method and accesses a time dict key (which is printed on new line each iteration).
- changed it to 20 second interval to make sure I am not running into rate limit issues.

```
# code.py file 

import board
import busio
from digitalio import DigitalInOut
import adafruit_requests as requests
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
from adafruit_esp32spi import adafruit_esp32spi
import time

try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

esp32_cs = DigitalInOut(board.ESP_CS)
esp32_ready = DigitalInOut(board.ESP_BUSY)
esp32_reset = DigitalInOut(board.ESP_RESET)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

requests.set_socket(socket, esp)

if esp.status == adafruit_esp32spi.WL_IDLE_STATUS:
    print("ESP32 found and in idle mode")
print("Firmware vers.", esp.firmware_version)
print("MAC addr:", [hex(i) for i in esp.MAC_address])

print("Connecting to AP...")
while not esp.is_connected:
    try:
        esp.connect_AP(secrets["ssid"], secrets["password"])
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(esp.ssid, "utf-8"), "\tRSSI:", esp.rssi)
print("My IP address is", esp.pretty_ip(esp.ip_address))

URL = "http://nextconnect.riderta.com/Arrivals.aspx/getStopTimes"

hdrs = {'Content-Type': 'application/json',
 'Accept': 'application/json',
 'Accept-Encoding': 'gzip, deflate',
 'Host': 'nextconnect.riderta.com',
 'Referer': 'http://nextconnect.riderta.com/LiveDepartureTimes',
 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.102 Safari/537.36'
 }

params = {'routeID': 139, 'directionID': 3, 'stopID': 13363, 'tpID': 0, 'useArrivalTimes': 'false'}
secs = 0
while True:
    print(secs, "Fetch start ", end = "")
    r = requests.post(URL,json=params,headers=hdrs)
    print("Fetch end")
    print(r.json()['d']['updateTime'])
    time.sleep(20)
    secs += 1
```

It connects and works great. It ran for about 6 hours and on the 1168th attempt it hung and USB unmounted. Last I saw on my serial console below.

```
....
1167 Fetch start Fetch end
6:28
1168 Fetch start
```

dsohrabian, 2022-05-18:
I'm fairly certain now that ESP32SPI isn't the issue and the problem lies with the matrix module and object in adafruit_matrixportal.
Closing here and continuing troubleshooting in more related open issue https://github.com/adafruit/circuitpython/issues/6205

closed 2022-05-18 by dsohrabian