adafruit/Adafruit_CircuitPython_ESP32SPI #174: 800 bps when serving webpage with WSGIServer

rivques, 2022-09-10:
Hi! I've been messing around with a Metro M4 Airlift Lite and hit an interesting snag when trying to serve a webpage using Adafruit's `esp32spi_wsgiserver` and `adafruit_wsgi.wsgi_app`. (Code is [here](https://github.com/rivques/CircuitPython/blob/master/internet.py) for those interested.) When serving the webpage, I was getting only about 800 *bits* per second, slow enough to cause Chrome to have issues recognizing the format of the page as HTML because (I assume) the `<!DOCTYPE html>` did not arrive quickly enough. After digging through `debug=True` logs and the various modules of `adafruit_esp32spi`, it seems like the issue is that `WSGIServer.finish_response` is sending data to `ESP_SPIcontrol.socket_write` one byte at a time. This bypasses `socket_write`'s built-in chunking and seems to cause a lot of extra overhead, possibly due to #108. [To fix this I use a slightly modified version](https://github.com/rivques/CircuitPython/blob/master/lib/my_wsgi.py) of `esp32spi_wsgiserver` where I `socket_write` the entire response at once if possible instead of looping over every char in the response and writing it individually. This increased data rate to something like 178 kbps. Anyways, my questions are:
1. Am I missing something that would cause this to be a non-issue?
1. Is this a known issue and if so, is there another workaround?
2. Would this problem be fixed by #108 being resolved?
3. Is my workaround a reasonable one, or does it have pitfalls I can't see? If so, should I put it in a PR to this repo?

dhalbert, 2022-09-10:
My intention in the near future is to remove  `esp32spi_wsgiserver` entirely and recommend that people use generic server implementations that use `adafruit_requests`, of which there are several. This makes the library smaller and allows us to write better server implementations that can be used in more circumstances. The PR to do that is #167, which is stalled for other reasons.

dhalbert, 2022-09-10:
Your fix sounds perfectly reasonable for now, and we can get it in for posterity.

PR #175 (2022-09-11): Close #174 by sending entire webserver response to ESP32 in one go if possible
This fixes #174, and gets my connection speed on a Metro M4 Airlift Lite from about 800 bps to something like 178 kpbs when serving a webpage, though your mileage may vary. For some reason `finish_response` would be called an empty array after the initial call with the intended response string, so I've left the original handling logic behind some type checks if that array ends up being non-empty. It's very possible I've misunderstood how WSGI is working here and more needs to be done with this, but it seems to handl [... cut]
commit 6f2b40e1 (2022-09-11): close #174
by sending entire webserver response to esp32 in one go if possible
commit ee6bfcf9 (2022-09-11): Merge pull request #175 from rivques/main

Close #174 by sending entire webserver response to ESP32 in one go if possible
closed 2022-09-11 by dhalbert