adafruit/Adafruit_CircuitPython_ESP32SPI #177: Feather M4 + AirLift feather  wing connect to WIFI error

smoczyna, 2023-02-06:
I copied the code here to my feather M4 and try to connect to my local WIFI.
Here is the effect:

>>> %Run -c $EDITOR_CONTENT
ESP32 SPI webclient test
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.2.2\x00')
MAC addr: ['0x5c', '0x12', '0x4a', '0x95', '0xab', '0x34']
	jama		RSSI: -37
	jama		RSSI: -68
	Horizon Wi-Free		RSSI: -73
	jama		RSSI: -74
	eir_WiFi		RSSI: -78
	jama		RSSI: -85
	HUAWEI_B628_1C47		RSSI: -92
Connecting to AP...
could not connect to AP, retrying:  ('No such ssid', b'jama')

my local wifi is jama, not b'jama and I cannot figure out where is this " b' " being added, nothing helps.

LAWBear, 2023-02-06:
Similar behavior to my efforts to manage GPC sentences:
b'$GNRMC,182606.000,A,3026.0536,N,095$G026.0536,N,09525.7007,W,2,11,0.92,87.3,M,-23.8,M,,*76'
LA

anecdata, 2023-02-06:
`b'something'` is just the Python syntax for `bytes` or a `bytearray`. For ASCII text, it looks as you see with the SSID. It's not a bug or anything to be concerned about. A lot of stuff goes over the network as bytes. If you want to display bytes as a string, there are various ways, like: `b'myssid'.decode()`.

If retrying doesn't help get beyond the "could not connect" message, then something else may be going on. I'd suggest hopping on the Adafruit Discord for support: https://adafru.it/discord

smoczyna, 2023-02-09:
indeed, it looks like network issue rather than code. However your airlift feather seems to be quite fossy, it doesn't connect throwing quite stupid error like te one above when other esp32s connect just fine.

closed 2023-02-09 by smoczyna