adafruit/Adafruit_CircuitPython_ESP32SPI #180: Unable to connect to WiFi with duplicate SSIDs

banksfinn, 2023-08-15:
### Problem
Currently, I'm unable to connect to my wifi network due to having multiple access points with duplicate SSIDs. There is no method for explicitly stating the BSSID or otherwise filtering down the SSIDs to prevent this issue.

### Tested
This was tested on an Adafruit Matrix Portal, using the guide listed here: https://learn.adafruit.com/adafruit-matrixportal-m4/internet-connect

This was first tested by the direct `esp.connect_AP`, then subsequently tested out using WifiManager.

### Solution

I'd love the ability to either explicitly state the BSSID, or default to connecting to the closest SSID.

I've made a branch for the WifiManager C++ code (here: https://github.com/banksfinn/WiFiManager), but realized this will need to be 2 part solution. This is the relevant thread I worked from: https://github.com/tzapu/WiFiManager/issues/221

Does anyone have any guidance of next steps for this repo?

### Hardware
Adafruit Matrix Portal
Wifi Mesh System

anecdata, 2023-08-16:
Connecting by BSSID as you note isn't currently implemented in this library like it is for native [wifi](https://docs.circuitpython.org/en/latest/shared-bindings/wifi/index.html#wifi.Radio.connect).

The NINA firmware will already try to connect to the best access point with the given SSID. This is done by the underlying Espressif esp-idf [call](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-reconnect).

My reading for what would be needed for ESP32SPI to connect using the BSSID of a particular AP...

This library's connect ([wifi_set_passphrase](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/blob/c8bb6258a0444c31ab5ae69e3af1fffbd7761a99/adafruit_esp32spi/adafruit_esp32spi.py#L470)) function would need the bssid parameter added.

The NINA firmware would need to be updated with the bssid parameter added to the [setPassPhrase](https://github.com/adafruit/nina-fw/blob/dffbd447bd413aeec5d8a0abf275dccca72ac2c7/main/CommandHandler.cpp#L67C5-L67C18) function.

However, the Arduino function called by NINA, [WiFi.begin](https://reference.arduino.cc/reference/en/libraries/wifi/wifi.begin/), doesn't appear to have that feature, so it might have to be re-created using Espressif esp-idf wifi calls. WiFi.begin does have the feature to connect to an open SSID by [channel](https://github.com/adafruit/nina-fw/blob/dffbd447bd413aeec5d8a0abf275dccca72ac2c7/arduino/libraries/WiFi/src/WiFi.cpp#L257).

trezm, 2025-05-27:
I see this issue is quite old, but wanted to add that I see the exact same issue. Here's the printout from the example wifi running:

```
ESP32 SPI webclient test
ESP32 found and in idle mode
Firmware vers. 1.2.2
MAC addr: <redacted>
        Daedalus                RSSI: -56
        Daedalus-guest          RSSI: -56
        Daedalus                RSSI: -60
        Daedalus-guest          RSSI: -61
        Daedalus                RSSI: -63
        Daedalus-guest          RSSI: -63
Connecting to AP...
could not connect to AP, retrying:  ('No such ssid', b'Daedalus')
```