adafruit/Adafruit_CircuitPython_ESP32SPI #190: Readme dependencies out of date

becca01s, 2024-02-27:
I was running a lab today and ran into a lot of issues students were having. Tracked it down to the most recent update (7.0) having a dependency on https://github.com/adafruit/Adafruit_CircuitPython_ConnectionManager Connection Manager. Currently I can't really see if this is listed anywhere, so its unclear where to point people to install. Am I missing something? Or is it just a link on the Readme.md that needs updating? For what its worth, installing the appropriate mpx connection manager in the /lib folder seems to work.

mMerlin, 2024-02-27:
Don't know about the readme, but I encounterd that issue doing some testing. The fix is:

```sh
circup install adafruit_connection_manager
```

Assuming you have cirup installed. Something like:

```sh
pipenv install --dev circup
```

FoamyGuy, 2024-02-28:
This is now listed in the dependencies section of the readme from #191. Thank you for reporting.

closed 2024-02-28 by FoamyGuy