adafruit/Adafruit_CircuitPython_ESP32SPI #210: Timed out waiting for SPI char

igorky, 2024-08-07:
Hi,
I have strange issue with code running on Adafruit MatrixPortal M4.
At some point of time it just stopped working. Maybe a year or so there were no problems at all.
All I see it's the 'Timed out waiting for SPI char' when I do a call via requests.get to get a json.
I upgraded to the latest CircuitPython 9.1.1 and Nina 1.7.7 with no luck to solve it.

Here is debug log:
`
*** Get host by name
*** Get socket
Allocated socket #0
*** Socket connect mode 2
*** Open socket to hostname 443 2
*** Closing socket #0
TimeoutError
ESP32 not responding
Timed out waiting for SPI char
`

What else can I check?

FoamyGuy, 2024-08-07:
Are you able to share the code.py file that you're attempting to run? 

I don't know of any specific cause for that error message off the top of my head, but if you can share the code I can try to run it and see if I'm able to replicate the same issue in order to look further into it.

T-Mosher, 2024-08-07:
I'm not entirely sure why this was posted here instead of on the Forum first. The repo is for reporting software bugs.

igorky, 2024-08-07:
Thank you, I will appreciate your help.
[code.zip](https://github.com/user-attachments/files/16533161/code.zip)

igorky, 2024-08-07:
> I'm not entirely sure why this was posted here instead of on the Forum first. The repo is mostly for reporting software bugs. The issue here could be a simple wiring failure, since the project was known to be working previously.
> 
> The forum has customer service techs to help diagnose issues.

There is no wiring, it's just one board and a display attached to it with no modifications. I am sure it's software issue, however I could not find it.

felker, 2025-04-24:
@igorky did you ever solve this?