adafruit/Adafruit_CircuitPython_ESP32SPI #214: Repeated open and close socket for UDP fails: OSError: Could not start server

larry5b6, 2024-10-17:
The issue came up when trying to send a UDP message to several destinations, but the simplified example attached demonstrates the issue with only one destination. Multiple connect - send - close cycles eventually cause a 'Could not start server' error. Environment is a pyportal running 9.1.4. 
[udpIssue.zip](https://github.com/user-attachments/files/17413300/udpIssue.zip)

anecdata, 2024-10-17:
Does the exception occur in a repeatable way, like after a certain number of iterations or a certain amount of time? I don't see this behavior in my own clients.

I noticed that the version of NINA firmware is old, you could try the latest.

If you catch the exception and retry, does it recover?

anecdata, 2024-10-18:
Hopefully you can get the NINA firmware updated and see if that improves things. I'm afk with just phone now, but I can do some testing next week when I'm back if the issue persists.

larry5b6, 2024-10-18:
Issue resolved by upgrading ESP32 firmware to 1.7.7. Case closed

closed 2024-10-18 by larry5b6